hardbyte/python-can #1263: Etas interface failed

shinhwap, 2022-02-23:
### Describe the bug
Hello, I'm trying using ETAS ES581.3 for CAN interface
I've installed BOA Distribution Package 16.12.2021 in my computer
I try to create EtasBus or run detect search available configs
But it will go to ValueError in etas __init__.py line 25
`CSI_CreateProtocolTree(ctypes.c_char_p(b""), nodeRange, ctypes.byref(self.tree))`

Please check the problem or the correct installation for ETAS interface

### To Reproduce
Here is my code
```
import can
from can.interfaces.etas import EtasBus
def main():
    configs = can.interface.detect_available_configs(interfaces="etas")
    for c in configs:
        print(c)
    bus = EtasBus(channel='ETAS://USB/ES581:45239', bitrate=500000, fd=False, data_bitrate=2000000)
    print(bus)
```

### Additional context

OS and version: Windows 10
Python version: 3.7.4
python-can version: 4.0.0
python-can interface/s (if applicable): ETAS 581.3

<details><summary>Traceback and logs</summary>

```
Traceback (most recent call last):
  File "C:\Users\A50044\AppData\Local\Programs\Python\Python37-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Users\A50044\AppData\Local\Programs\Python\Python37-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "E:\bls2\venv\lib\site-packages\can\logger.py", line 224, in <module>
    main()
  File "E:\bls2\venv\lib\site-packages\can\logger.py", line 194, in main
    bus = _create_bus(results, can_filters=_parse_filters(results), **additional_config)
  File "E:\bls2\venv\lib\site-packages\can\logger.py", line 111, in _create_bus
    return Bus(parsed_args.channel, **config)  # type: ignore
  File "E:\bls2\venv\lib\site-packages\can\interface.py", line 120, in __new__
    bus = cls(channel, *args, **kwargs)
  File "E:\bls2\venv\lib\site-packages\can\interfaces\etas\__init__.py", line 25, in __init__
    CSI_CreateProtocolTree(ctypes.c_char_p(b""), nodeRange, ctypes.byref(self.tree))
ValueError: Procedure probably called with too many arguments (16 bytes in excess)
```
</details>

shinhwap, 2022-02-24:
I found the problem
I'm using 32 bit python
It should use ctypes.CDLL in boa.py
```
    # try to load libraries with hardcoded paths
    if ctypes.sizeof(ctypes.c_voidp) == 4:
        # 32 bit
        path = "C:/Program Files (x86)/ETAS/BOA_V2/Bin/Win32/Dll/Framework/"
        _csi = ctypes.CDLL("dll-csiBind")
        _oci = ctypes.CDLL("dll-ocdProxy")
    elif ctypes.sizeof(ctypes.c_voidp) == 8:
        # 64 bit
        path = "C:/Program Files/ETAS/BOA_V2/Bin/x64/Dll/Framework/"
        _csi = ctypes.windll.LoadLibrary(path + "dll-csiBind")
        _oci = ctypes.windll.LoadLibrary(path + "dll-ocdProxy")
```

But after I load correct dll
I still get this error
```
Traceback (most recent call last):
  File "test1.py", line 37, in <module>
    main()
  File "test1.py", line 13, in main
    bus = EtasBus(channel='ETAS://USB/ES581:45239/CAN:1', bitrate=500000)
  File "E:\bls2\venv\lib\site-packages\can\interfaces\etas\__init__.py", line 34, in __init__
    ctypes.byref(self.ctrl),
  File "E:\bls2\venv\lib\site-packages\can\interfaces\etas\boa.py", line 30, in errcheck_init
    raise CanInitializationError(f"{func.__name__} failed with error 0x{result:X}")
can.exceptions.CanInitializationError: OCI_CreateCANControllerNoSearch failed with error 0x80001000
```

I test my E581.3 device with busmaster and I confirm that device works.

zariiii9003, 2022-02-24:
@felixn can you take a look?