hardbyte/python-can #1267: Add support for J1939 message decoding in Viewer

simonkerscher, 2022-03-01:
### Is your feature request related to a problem? Please describe.
J1939 messages include a few features, which are currently not supported by the viewer.
1. Messages might have 24-bit fields, e.g. PGN 0xF029 (slope sensor information 2)
2. Data fields have a scaling and an offset
3. All messages with the same PGN (middle 2 bytes of cob-id) can be decoded the same way, outer bytes only define priority and sending device

### Describe the solution you'd like
1. Introduce another option, e. g. `J`. Example usage `python3 -m can.viewer ... -d "0CF02980::<JJxx"` would decode message 0CF02980 with bytes 0-2 and bytes 3-5 as 24-bit data, and use byte 6-7 as padding byte.
2. Introduce an option to apply an offset to a decoded message, e.g. `,`. Example usage `python3 -m can.viewer ... -d "0CF02980:<JJxx:32768,-250:32768:-250"` would divide bytes 0-2 and 3-5 by 32768 and apply an offset of -250.
3. Introduce additional syntax similar to the current state of `-f`, where all messages that fit the filter get decoded in the same way. However the obvious symbol `:` already has another use for `-d`.

### Additional context
Items 1 and 2 have higher priority for me, and I'm happy to implement those if you agree that this would be a useful addition

zariiii9003, 2022-03-01:
Hi, python-can does not implement protocols (yet?). But there is [this library](https://github.com/juergenH87/python-can-j1939) which builds on top of python-can and seems to be actively developed.

hexsecs, 2022-08-17:
@simonkerscher there are some pretty good tools out there for this. Some of them are listed here...
https://github.com/iDoka/awesome-canbus#j1939-tools

hardbyte, 2022-08-18:
`python-can` actually implemented J1939 in the past but we deliberately made the decision to focus on raw CAN and leave higher level protocols for libraries and applications (hoping that they could utilize python-can). See https://github.com/hardbyte/python-can/issues/65

closed 2022-08-18 by hardbyte