hardbyte/python-can #1275: Error with Multicast IP Interface

Gudoghe, 2022-03-08:
Hello everyone, I'm fairly new to the python-can environment and I'm trying to get the basic example for the Multicast IP Interface. 
I've installed the python-can module using pip, I've copied the example for this specific CAN connection from the documentation ([https://python-can.readthedocs.io/en/develop/interfaces/udp_multicast.html#bus-class-documentation](https://python-can.readthedocs.io/en/develop/interfaces/udp_multicast.html#bus-class-documentation)), but I seem to get an error that has to do with the bus.py file in the interface installation folder. 

The error I'm getting is the following:

```
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/can/interfaces/udp_multicast/bus.py", line 241, in _create_socket    sock.setsockopt(socket.SOL_SOCKET, SO_TIMESTAMPNS, 1)
OSError: [Errno 42] Protocol not available
```

This feels really weird since the same function is called just a few lines prior in the bus.py file, but it does not raise an error.

I'm on MacOS using python3.10 and python-can version 4.0.0

Is there something I'm doing wrong? Any help would be really appreciated.

felixdivo, 2022-03-09:
Side note: Now that I look at it, maybe this line should also be changed to say only allow multiple addresses, not ports: https://github.com/hardbyte/python-can/blob/develop/can/interfaces/udp_multicast/bus.py#L237-L238

felixdivo, 2022-03-09:
I found [this SO article](https://stackoverflow.com/questions/50959310/enabling-udp-packet-timestamps-on-macos-with-scm-timestamp) that says in the question:

> I am aware that the socket implementation in macOS follows POSIX (and as such do not expect `SO_TIMESTAMPNS` to work)

Sounds plausible to me. Maybe we should allow for that to fail. We would have to change [this](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/udp_multicast/bus.py#L323-L337) thing only. It was more or less copied  from the socketcan driver (see [setup](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/socketcan/socketcan.py#L684-L689) and [receiving](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/socketcan/socketcan.py#L538-L550)), since it is guaranteed to work from Linux 2.6.22 on. Apparently though, not for macOS.

The implementation would be easy, e.g. use what was present in socketcan before [this change](https://github.com/hardbyte/python-can/pull/1015/files#diff-251cfd54950cdf7c440a19de22c9d61623a076f388ba71bde1d326f0bbc69ca7L533-L537), depending whether the `SO_TIMESTAMPNS` setting failed or not.

felixdivo, 2022-03-09:
@Gudoghe could you test a patch to confirm it being fixed?

Gudoghe, 2022-03-09:
> @Gudoghe could you test a patch to confirm it being fixed?

Yes absolutely, thank you for taking the time!

felixdivo, 2022-03-09:
Let's continue in #1278

felixdivo, 2022-04-16:
I's not clear whether https://github.com/hardbyte/python-can/pull/1278 solved the issue but I'll stop there

PR #1278 (2022-04-29): Fix timestamp handling in udp_multicast on macOS
Fix #1275.
closed 2022-04-29 by hardbyte