hardbyte/python-can #1284: Vector interface doesnt send msg with correct data or id (hardware config-> driver status -> init is missing)

matze4949, 2022-03-15:
### Describe the bug
When i run the following script alone I always receive messages without any data or id:
```import can
import time

mainCan = can.interface.Bus(interface='vector',channel=0,app_name='CANoe')
recvCan = can.interface.Bus(interface='vector',channel=1,app_name='CANoe')

msg = can.Message(arbitration_id=0x23,data=[0x12,0x13])

mainCan.send(msg)
print(msg)
recvMsg = recvCan.recv()
print(recvMsg)

time.sleep(15)
mainCan.shutdown()
recvCan.shutdown()
```
I get the following result and driver status out of hardware control:
```
Timestamp:        0.000000    ID: 00000023    X Rx                DL:  2    12 13
Timestamp: 1647341231.680868        ID: 0000    S Rx E              DL:  0                                Channel: 1
```
[image]

### With CANoe in background
When I run a CANoe File in the background i can receive the correct message (code is the same).
Result:
```
Timestamp:        0.000000    ID: 00000023    X Rx                DL:  2    12 13
Timestamp: 1647341634.418287    ID: 00000023    X Rx                DL:  2    12 13                       Channel: 1
```
[image]

there is only a small diffrence in the driver status, when running CANoe in the background init is shown in the Hardware Config

### Additional context

OS and version: Win10 1909
Python version: Py3.8.5
python-can version: 4.0.0
python-can interface/s (if applicable): Vector 1630A

zariiii9003, 2022-03-15:
It works fine when i test with virtual channels. 

Only the first app on the bus gets Init access. Maybe there's another active application in your case. Your Baudrate is also different between the screenshots.

Maybe you received a message from another app?

matze4949, 2022-03-15:
Now i get the init access but the receiving message is still empty or 0x00

zariiii9003, 2022-03-15:
Can you edit the can.interfaces.vector.canlib.py file like this?

This should be at line 479:
```python3
    def _recv_can(self) -> Optional[Message]:
        xl_event = xlclass.XLevent()
        event_count = ctypes.c_uint(1)
        self.xldriver.xlReceive(self.port_handle, event_count, xl_event)
        print(xldriver.xlGetEventString(xl_event).decode())
        if xl_event.tag != xldefine.XL_EventTags.XL_RECEIVE_MSG:
            self.handle_can_event(xl_event)
            return None
```

I added a print statement to see if there are errors on the bus.

zariiii9003, 2022-03-15:
I just noticed that you received an error frame here
```
Timestamp:        0.000000    ID: 00000023    X Rx                DL:  2    12 13
Timestamp: 1647341231.680868        ID: 0000    S Rx E              DL:  0                                Channel: 1
```
Do you have the necessary CAN Termination between Channel1 and Channel2?

matze4949, 2022-03-16:
with the print command i get this:
`RX_MSG c=1, t=495096619008, type=0000,  ERROR_FRAME tid=00`

For the first tests with python-can I just build a loop with two channels, between the 2 cabels there is a USA PATENT 519906 Gender Changer, thats all.

matze4949, 2022-03-16:
oh man.... just forgot about that part.

its fixed thank you

zariiii9003, 2022-03-16:
yes, bitrate is correct

closed 2022-03-17 by matze4949