hardbyte/python-can #1285: Unable to send periodic messages with IXXAT interface as of 4.0.0.

thshan, 2022-03-16:
### Description:
While attempting to set up a periodic task to synchronously write/read PDOs using an IXXAT USB-to-CAN and  the [canopen](https://github.com/christiansandberg/canopen) package, an AttributeError is generated due to the `IXXATBus` object missing `_periodic_tasks` (see: traceback below).

### Expected behavior
On v3.3.4, it is possible to use `node.rpdo[pdo].start` or `network.sync.start` from the `canopen` package (on its latest 2.0.0 release) without any errors being generated. Calling the constructor does not resolve the issue on V4.0.0, however, manually setting `bus._periodic_tasks = []` functions as a workaround. 

### Additional context

OS and version: Windows 10 Professional (10.0.19044 Build 19044)
Python version: Python 3.9.10, also confirmed with Python 3.10.3
python-can version: 4.0.0
python-can interface/s (if applicable): IXXAT. 

<details><summary>Traceback and logs</summary>

```

Traceback (most recent call last):
 line 78, in <module>
    main()
  line 54, in main
    network.start_nodes(net, 0.5)
  start_nodes
    node.rpdo[pdo].start(sync_time)
 line 449, in start
    self._task = self.pdo_node.network.send_periodic(
  "...\canopen\network.py", line 220, in send_periodic
    return PeriodicMessageTask(can_id, data, period, self.bus, remote)
 "...\canopen\network.py", line 294, in __init__
    self._start()
 "...\canopen\network.py", line 297, in _start
    self._task = self.bus.send_periodic(self.msg, self.period)
  "...\python_can\can\bus.py", line 239, in send_periodic
    periodic_tasks = self._periodic_tasks
AttributeError: 'IXXATBus' object has no attribute '_periodic_tasks'
```

Bus attributes on 4.0.0:
['RECV_LOGGING_LEVEL', '__abstractmethods__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__enter__', '__eq__', '__exit__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_abc_impl', '_apply_filters', '_detect_available_configs', '_matches_filters', '_recv_internal', '_send_periodic_internal', 'bus', 'channel_info', 'fileno', 'filters', 'flush_tx_buffer', 'recv', 'send', 'send_periodic', 'set_filters', 'shutdown', 'state', 'stop_all_periodic_tasks']

Bus attributes on 3.3.4:
['CHANNEL_BITRATES', 'RECV_LOGGING_LEVEL', '_IXXATBus__set_filters_has_been_called', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__enter__', '__eq__', '__exit__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__lt__', '__metaclass__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_apply_filters', '_channel_capabilities', '_channel_handle', '_control_handle', '_detect_available_configs', '_device_handle', '_device_info', '_inWaiting', '_matches_filters', '_message', '_payload', '_periodic_tasks', '_receive_own_messages', '_recv_internal', '_scheduler', '_scheduler_resolution', '_send_periodic_internal', '_tick_resolution', 'channel', 'channel_info', 'filters', 'flush_tx_buffer', 'recv', 'send', 'send_periodic', 'set_filters', 'shutdown', 'state', 'stop_all_periodic_tasks']

</details>

zariiii9003, 2022-03-18:
Could you try this branch?
https://github.com/zariiii9003/python-can/tree/fix_1285

thshan, 2022-03-18:
> Could you try this branch? https://github.com/zariiii9003/python-can/tree/fix_1285

Hi zariiii9003. No change on that branch. I have, however, added some updated details in the OP after a bit of investigation.

zariiii9003, 2022-03-18:
Can you show your traceback with my branch?