hardbyte/python-can #1287: Possibility to add partials as listeners to the notifier

tamenol, 2022-03-28:
### Is your feature request related to a problem? Please describe.

At the moment it's not possible to use `functools.partial`s as callback functions. `ValueError` gets raised on the line `self.listeners.remove(listener)` when trying to remove the listener since `functools.partial` doesn't have `__eq__`.

### Describe the solution you'd like

Change the `remove_listener` function a bit:

```python    
def remove_listener(self, listener: Listener) -> None:
        """Remove a listener from the notification list. This method
        throws an exception if the given listener is not part of the
        stored listeners.

        :param listener: Listener to be removed from the list to be notified
        :raises ValueError: if `listener` was never added to this notifier
        """
        if isinstance(listener, partial):
            for l in self.listeners:
                if isinstance(l, partial) and l.func == listener.func and l.args == listener.args and l.keywords == listener.keywords:
                    self.listeners.remove(l)

        else:
            self.listeners.remove(listener)
```

Or something in these lines, this was just quickly thrown together to test it out.

### Describe alternatives you've considered

- Implement this in your own application above this module
- Inherit `functools.partial` and implement `__eq__`

### Additional context

zariiii9003, 2022-03-29:
I did not test this but you could probably check for identity with the `is` operator without specifically mentioning `functools.partial`.

```python3
try:
    self.listeners.remove(listener)
except AttributeError:
    for item in self.listeners:
        if item is listener:
            self.listeners.remove(item)
            break
    else:
        raise ValueError(f"{listener} not in list.")
```

tamenol, 2022-03-30:
@zariiii9003 the failed `remove` instruction throws a `ValueError`. Also `is` does not work, since `partial` does not have the implementation function that can check this.

I did not know about the `for - if break - else` construct. Thanks for this. Is quite handy

zariiii9003, 2022-03-30:
Can you post the traceback of your error with the `__eq__` method or a code sample to reproduce?

tamenol, 2022-03-30:
@zariiii9003 I made a mistake when programming my application and the whole premise of this issue is wrong. My apologies.

I needed to assign the `partial` to a variable to make it work.

This works:
```python
def _callback_msg(port: int, msg: can.Message):
    print(f"{port} - {msg.data}")

callback_partial = partial(_callback_msg, 1)
listeners = []
listeners.append(callback_partial)

print(callback_partial in listeners)
# --> True
```

This is how I did it earlier (and wrongly):
```python
def _callback_msg(port: int, msg: can.Message):
    print(f"{port} - {msg.data}")

listeners = []
listeners.append(partial(_callback_msg, 1))

print(partial(_callback_msg, 1) in listeners)
# -> False
```

So this can be closed. Sorry for taking away your time.

zariiii9003, 2022-03-30:
No problem, i was wondering why i didn't have any issues removing partial from a list.

closed 2022-03-30 by zariiii9003