hardbyte/python-can #1288: usb2can device

superlogan72, 2022-04-07:
Hello i have buy a usb2can by inno-maker but I'm not able get it work 
I'm under windows 10 and the device is well recognized by the system as per attached picture

[image]
but I'm not able to configure it with python-can
Can you give me some idea or just suggest me a working usb2can device with this library
Thanks

Muteskinto, 2022-04-08:
Hi there,
I have (I assume, since it is recognized by Windows in the same way) the same Device.

I got it to work with python-can using the "gs_usb" interface Module.
Follow the explanations in the [docs](https://python-can.readthedocs.io/en/master/interfaces/gs_usb.html).

I had to manually download libusb-1.0.x for Windows and move the .dll to \Windows\System32 and then change the driver with Zadig to libusb-win32. It will say the driver is libusb0 after that, but down worry about it.

Have fun!

superlogan72, 2022-04-08:
thanks for your reply but i don't know how to setup the bus conf string
bus = can.Bus(bustype="gs_usb", channel=dev.product, bus=dev.bus, address=dev.address, bitrate=250000)
what i write in dev.bus - dev.product dev.address?
Thanks
Luca

superlogan72, 2022-04-08:
i have done it but i get the error:
Exception "unhandled usb.core.USBError"
[Errno 13] Access denied (insufficient permissions)
File: C:\Users\Luca\AppData\Local\Programs\Python\Python38\lib\site-packages\usb\backend\libusb1.py, Line: 604
I'm using win10 64bit

superlogan72, 2022-04-08:
I have redo all and now it works.
Many thanks

superlogan72, 2022-04-11:
Sorry again i try a simple program but i get this error
Can you suggest something?

nodo_luca = BaseNode402(1, 'EBS.eds')
network = canopen.Network()
network.add_node(nodo_luca)
network.connect(bustype="gs_usb", channel=dev.product, bus=dev.bus, address=dev.address, bitrate=500000)
network.scanner.search()
time.sleep(0.05)
for node_id in network.scanner.nodes:
print("Found node %d!" % node_id)
#here i get Found node 1!
node.setup_402_state_machine()
#i get:
Exception "unhandled AssertionError"
OPERATIONAL
File: C:\Users\Luca\AppData\Local\Programs\Python\Python38\lib\site-packages\canopen\profiles\p402.py, Line: 239

gerryqd, 2023-03-23:
Maybe you can try to replace the driver to "libusbK" instead of "libusb-win32" and then have a try.
I got issues with libusb-win32, but works well if using the libusbK.