hardbyte/python-can #1291: AsyncBufferedReader misses messages

pthomas, 2022-04-14:
Hello,

I'm using can.AsyncBufferedReader(), and it seems if there is too much time time between reader.get_message() calls then messages are lost. They are still seen by wireshark, but not from reader.get_message(). Is this the expected behavior? Is there anything that can be done about this?

Even very small sleeps like time.sleep(0.001) will cause this issue.

OS and version: Ubuntu
Kernel: 5.4.0-107-generic
Python version: 3.8.10
python-can version: 4.0.0
python-can interface/s (if applicable): socketcan w/ peak_usb

Any help would be appreciated.

-Paul

MohamedAboushnief, 2023-06-22:
@pthomas  did you solve this issue?

zariiii9003, 2023-06-22:
Please provide a code example. Using the blocking `time.sleep(0.001)` with asyncio looks wrong to me.