hardbyte/python-can #1292: Error when trying to initialize isCAN dongle

Jeffmagma, 2022-04-14:
### Describe the bug
When I try to set up the isCAN bus the same way I did with PCAN (tested and working for me), I get an error:
```
File "C:\Users\Jeffmagma\Documents\pythontest.py", line 182, in <module>
    main()
  File "C:\Users\Jeffmagma\Documents\pythontest.py", line 168, in main
    bus = can.Bus(interface='iscan', bitrate=250000, channel=0)
  File "C:\Users\Jeffmagma\AppData\Roaming\Python\Python39\site-packages\can\interface.py", line 120, in __new__
    bus = cls(channel, *args, **kwargs)
  File "C:\Users\Jeffmagma\AppData\Roaming\Python\Python39\site-packages\can\interfaces\iscan.py", line 106, in __init__
    iscan.isCAN_DeviceInitEx(self.channel, self.BAUDRATES[bitrate])
  File "C:\Users\Jeffmagma\AppData\Roaming\Python\Python39\site-packages\can\interfaces\iscan.py", line 35, in check_status_initialization
    raise IscanInitializationError(function, result, arguments)
  File "C:\Users\Jeffmagma\AppData\Roaming\Python\Python39\site-packages\can\interfaces\iscan.py", line 195, in __init__
    f"Function {self.function.__name__} failed{description}",
AttributeError: 'IscanInitializationError' object has no attribute 'function'
```

### To Reproduce
This can be reproduced with the simple python code:
```python
import can
bus = can.Bus(interface='iscan', channel=0)
```

### Expected behavior
I expect the error not to occur and my program to continue to run as usual

### Additional context

OS and version: Windows 10 21H2
Python version: 3.10
python-can version: 4.0.0
python-can interface/s (if applicable): working with PCAN, not working with isCAN

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

zariiii9003, 2022-04-18:
Can you try again with this branch? https://github.com/zariiii9003/python-can/tree/fix_1292
You should get a meaningful error message now.

Jeffmagma, 2022-04-19:
Seems like that fixed the issue, thanks

closed 2022-04-19 by Jeffmagma

Jeffmagma, 2022-04-19:
Is there a forum or chat for this library? I have a question but don't know where to ask it. I just want to know how to get the source address that python-can uses to send its messages

PR #1293 (2022-04-20): Fix AttributeError in IscanError
Close #1292
commit ed6cd668 (2022-04-20): fix #1292 (#1293)