hardbyte/python-can #1294: CANalyst-II can't reopen device

MoscowskyAnton, 2022-04-15:
Hello! I use your library with CANalyst-II device and it works pretty well, thank you!
But I have one problem, first run of program is fully successful, but when I quit it and start again it throws error like:

```
    msg = self.bus.recv(1)
  File "/home/anton/.local/lib/python3.8/site-packages/can/bus.py", line 94, in recv
    msg, already_filtered = self._recv_internal(timeout=time_left)
  File "/home/anton/.local/lib/python3.8/site-packages/can/interfaces/canalystii.py", line 177, in _recv_internal
    self.poll_received_messages()
  File "/home/anton/.local/lib/python3.8/site-packages/can/interfaces/canalystii.py", line 157, in poll_received_messages
    (channel, raw_msg) for raw_msg in self.device.receive(channel)
  File "/home/anton/.local/lib/python3.8/site-packages/canalystii/device.py", line 259, in receive
    assert len(rx_data) % self.MESSAGE_BUF_LEN == 0
AssertionError
free(): double free detected in tcache 2
```
If I replug device everything is Ok again.  
In code it is looks like:
```python
self.bus = can.interface.Bus(bustype="canalystii", channel="0", bitrate=125000)
...
while True:
   msg = self.bus.recv(1)
...
self.bus.shutdown()
```

OS and version: Kubuntu 20.04
Python version: 3.8.10
python-can version: 4.0.0
python-can interface/s: canalystii==0.1

Any ideas?

projectgus, 2022-04-27:
That's a bit weird, the assertion is failing because USB returned less bytes than expected.

Can you try printing the actual number of bytes before the assert, please? i.e.

```
@@ -256,6 +256,7 @@ class CanalystDevice(object):
         message_ep = CHANNEL_TO_MESSAGE_EP[channel]
         rx_data = self._dev.read(message_ep | 0x80, rx_buffer_size)
 
+        print(len(rx_data), rx_buffer_size)
         assert len(rx_data) % self.MESSAGE_BUF_LEN == 0
         num_buffers = len(rx_data) // self.MESSAGE_BUF_LEN
 
```

Maybe the canalystii backend needs to reset the device somehow when it reopens it, or flush any pending reads.

MoscowskyAnton, 2022-04-28:
Thanks for response!

It printed two lines before crush
`0 128
48 128`
And if I keep launching in increases
`0 10752
48 10752
`
I tried to add `flush_tx_buffer` after bus init, but this doesn't help
There is also in canalystii function `clear_rx_buffer` but how can I reach it from `can` class?

projectgus, 2022-05-01:
`MESSAGE_BUF_LEN` is 64 (each CAN message is 64 bytes long in the device's USB protocol). For some reason your device says it has messages available and then does a USB "short read" and only sends 48 bytes (less than a full message.)

i.e. the rx_buffer_size 128 is for 2 messages, 10752 is for 168 messages because the RX buffer is filling up.

Try `bus.device.clear_rx_buffer(0)` and see if you can start reading properly after this.

Otherwise you can try commenting the `assert` line. In your case, it will read 0 messages this time and maybe the next read will be correct? Although I don't know where the missing 16 bytes has gone...

MoscowskyAnton, 2022-05-04:
Clearing buffer doesn't help but commenting assert line almost does
Now it reconnects successfully several times, until throws 
```
    msg = self.bus.recv(1)
  File "/home/anton/.local/lib/python3.8/site-packages/can/bus.py", line 94, in recv
    msg, already_filtered = self._recv_internal(timeout=time_left)
  File "/home/anton/.local/lib/python3.8/site-packages/can/interfaces/canalystii.py", line 177, in _recv_internal
    self.poll_received_messages()
  File "/home/anton/.local/lib/python3.8/site-packages/can/interfaces/canalystii.py", line 157, in poll_received_messages
    (channel, raw_msg) for raw_msg in self.device.receive(channel)
  File "/home/anton/.local/lib/python3.8/site-packages/canalystii/device.py", line 257, in receive
    rx_data = self._dev.read(message_ep | 0x80, rx_buffer_size)
  File "/home/anton/.local/lib/python3.8/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/home/anton/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/home/anton/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/anton/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out
free(): double free detected in tcache 2
```
But seems it is hardware problem, as soon it is strongly correlate with blinking on a device I use.

Nareshkumar-g, 2023-02-07:
Hi,
I am using catalystii-II device to control my BLDC motor,
While running python_catalystii.py file , i am getting the following error 
  bus = can.interface.Bus(bustype='canalystii', channel=1, bitrate=500000)

  File ~\Anaconda3\lib\site-packages\can\util.py:369 in wrapper
    return f(*args, **kwargs)

  File ~\Anaconda3\lib\site-packages\can\interface.py:138 in __new__
    bus = cls(channel, **kwargs)

  File ~\Anaconda3\lib\site-packages\can\interfaces\canalystii.py:116 in __init__
    channel = msg.channel

AttributeError: 'NoneType' object has no attribute 'VCI_OpenDevice',

Can anyone give solution for this problem ?