hardbyte/python-can #1298: adding a generic CANalyst-II backend

mikisama, 2022-04-27:
Hi, guys.

Version 3.x of python-can worked well with my CANalyst-II. However, in version 4.x, I found that the backend of python-can for CANalyst-II has changed. My CANalyst-II can not work with python-can.

* As far as I know, the genuine CANalyst-II is manufactured by [ZLG](https://www.zlg.com/), but there are many cloned CANalyst-II devices on the market, whose manufacturers provide ZLG-compatible DLLs.

* Version 3.x of python-can can support CANalyst-II of different manufacturers by simply replacing the DLL.
* Version 4.x of python-can may only support CANalyst-II devices from Chuangxin Technology.

Would you consider adding a 'generic' CANalyst-II backend like the 3.x version?

zariiii9003, 2022-04-27:
@projectgus Could you take a look?

projectgus, 2022-04-27:
Oh wow, this is very interesting! @mikisama I did not actually know there was a "genuine" CANalyst-II from ZLG available for sale.

I did know that the Chuangxin Technology one came with a DLL that made it work with the ZLG CANPro Windows program, but in the Windows program it doesn't actually say "Canalyst-II" in the interface.

Here's all of the versions I can find:

1. Chuangxin Technology version. [Example](https://www.aliexpress.com/item/1005004146220418.html). Available on Taobao, Aliexpress, etc. This is the version I have.
2. Looks the same as 1 but doesn't say "Chuangxin Technology" on the case. [Example](https://detail.tmall.com/item.htm?id=542301692415). According to Taobao sellers this is still Chuangxin Technology version, but I haven't confirmed that the USB protocol & DLL are the same. Same device also available in silver.
3. [iTek Canalyst-II](https://item.taobao.com/item.htm?id=569520349016). I only saw this version today. I don't know anything about it.
4. ZLG [CANalyst-II+](https://www.zlg.com/index.php/can/can/product/id/42.html). This product doesn't seem to really exist, I can't find anywhere to buy it. The ZLG official Taobao store doesn't have it. They sell the "USBCAN" device instead, and [there is a PR to add support for the ZLG USBCAN](https://github.com/hardbyte/python-can/pull/1209).

It seems like there are two ways forward:

1. Revert #1127 and go back to using the DLL. This will remove some things fixed in the PR, but it will mean that other "Canalyst-II" versions can work by changing the DLL.
2. Find out what the difference in USB protocol is for the other Canalyst-II types, and add support in [the Python backend](https://pypi.org/project/canalystii/). I am interested to help with this, if there are not too many different clones.

@mikisama @zariiii9003 Can you share about the exact Canalyst-II that you have? A photo, where you bought it, when you bought it, etc?

mikisama, 2022-04-27:
> 
Hi, @projectgus 
Thanks for your reply.

I have this one [USBCAN-II](http://www.gcgd.net/prolist_26_48.html)

The stuff in this zip is a ZLG-compatible DLL and manual.
[Guangcheng Technology USBCAN Interface Function Library Instruction Manual.zip](https://github.com/hardbyte/python-can/files/8571883/Guangcheng.Technology.USBCAN.Interface.Function.Library.Instruction.Manual.zip)

projectgus, 2022-05-01:
Thanks, that's interesting. So the one you have is actually a "ZLG"USBCAN" compatible clone, not "Canalyst-II" exactly, but it all works with the same ControlCan DLL and the "CANPro" ZLG software?

The Shenyang Guangcheng Technology CAN protocol support in their documentation actually sounds better than the basic Canalyst-II protocol support which doesn't allow much detection of CAN bus errors. I am wondering if I should get one of these and write a pure Python backend for it as well. :)

I think #1209 by @keelung-yang will add support for this USBCAN device again, because this PR is using the DLL as well. It's interesting that USBCAN worked in the old versions using the DLL for canalystii backend though, I didn't know this and I don't think the maintainers knew this!

keelung-yang, 2022-05-01:
As discussed with supporters from ZLG, they may use incompatible libraries just for supporting Windows/Linux or CAN/CAN-FD, or even for the same product as asked by different clients.

We also discussed `Canalyst-II` in python-can v3.x, and they tell me I can't use it for our USBCANFD-100U/200U. That's why I develop a new back-end: Only for Linux. On Windows they use different APIs. There are many works need to be done for support both Linux and Windows.

In short, there are huge compatibility issues among ZLG official libraries. You can only expect compatible APIs to manipulate ZLG CAN devices, such as Kvaser/Vector/Intrepid..., only if ZLG decide to develop one. Or there're many many works need to be done for compatibility.

Nareshkumar-g, 2023-03-09:
I want to use python-can [canalst-II] in pyhton 2.7, what can i do?

mikisama, 2023-03-10:
> I want to use python-can [canalst-II] in pyhton 2.7, what can i do?

You can try install the old version
```
pip install python-can==3.3.4
```