hardbyte/python-can #1300: messages cannot display in bus when bus1 and bus2 use different channel

catmemo, 2022-04-28:
### Describe the bug
I have bus1 and bus2 as below:
```
bus1 = can.interface.Bus(bustype='vector', app_name=None, channel=0, bitrate=500000)
bus2 = can.interface.Bus(bustype='vector', app_name=None, channel=1, bitrate=500000)
```

I tried to send some messages by bus1 and print them as below:
```
for msg in bus1:
     print(msg)
```

However, the messages did not printed when bus1 and bus2 use different channel even the bus2 was not used.
And If I set both channel to 0 and it works.

### To Reproduce
```
import can
import cantools
import time

task = locals()
def send():

    bus1 = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=0, bitrate=500000)
    bus2 = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=1, bitrate=500000)
    db = cantools.database.load_file("xxx")
            # ...
    try:
        signals_data = {'BCS_VehSpdVD': 1, 'BCS_VehSpd': 240, 'BCS_VDCFaultSt': 0, 'BCS_TCSActiveSt': 0, 'BCS_TCSFaultSt': 0, 'BCS_EBDActiveSt': 0, 'BCS_EBDFaultSt': 0, 'BCS_ABSFaultSt': 0, 'BCS_ABSActiveSt': 0, 'BCS_VDCActiveSt': 0, 'BCS_2_A_MsgCounter': 0, 'BCS_2_A_Checksum': 85}
        db_message = db.get_message_by_name("GW_BCS_2_B")
        
        while True:
            
            data = db_message.encode(signals_data)
            message = can.Message(arbitration_id=0x260, data=data, is_extended_id=False)
##            if "test" in locals() or "test" in globals():
##                task["test"].stop(remove_task=True)
            print("OK")
            task["test"] = bus1.send_periodic(message, 0.01, duration=None, store_task=True)
##            assert isinstance(task["test"], can.CyclicSendTaskABC)
            
            if signals_data["BCS_VehSpd"] == 240:
                signals_data["BCS_VehSpd"] = 0
            else:
                signals_data["BCS_VehSpd"] = 240
                
            time.sleep(2)
            
            data = db_message.encode(signals_data)
            message = can.Message(arbitration_id=0x260, data=data, is_extended_id=False)
            task["test"].modify_data(message)

            for msg in bus1:
                print(msg)
            
            
    except can.CanError:
        print("Message NOT sent")

if __name__ == "__main__":
    send()
```

### Expected behavior

0x260 should be printed as well

### Additional context

OS and version: win10
Python version: 3.7.2
python-can version: latest
python-can interface/s (if applicable): No

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

zariiii9003, 2022-04-29:
Are you connected to a physical bus? Or is CANalyzer assigned to a virtual channel?
Use[ `receive_own_messages=True`](https://python-can.readthedocs.io/en/master/interfaces/vector.html#can.interfaces.vector.VectorBus) if you want to receive messages that you sent.

catmemo, 2022-04-29:
> Are you connected to a physical bus? Or is CANalyzer assigned to a virtual channel? Use[ `receive_own_messages=True`](https://python-can.readthedocs.io/en/master/interfaces/vector.html#can.interfaces.vector.VectorBus) if you want to receive messages that you sent.

I connected to a physical bus. 
OK, I'll try [receive_own_messages=True](https://python-can.readthedocs.io/en/master/interfaces/vector.html#can.interfaces.vector.VectorBus) and feedback here, thanks~

closed 2022-05-07 by catmemo