hardbyte/python-can #1304: Discover device needed

GiorgioSgl, 2022-05-03:
### Is your feature request related to a problem?
I need a function or something to discover the device connected to the can protocol and it's Id, I don't find any way to do it actually and neither suggestion

### Describe the solution you'd like
I'm thinking about the possibility to create a function to discover devices. I have no idea about how to do it actually.

### Describe alternatives you've considered
I have tried something like this but without success, and also broadcast manager handle only the sending of messages but not the reception wich will be useful.

```python
def get_leds_connected(initial_timeout=3, channel="can0"):
    leds_id = None

    with can.Bus(interface="socketcan",
                 channel=channel,
                 bitrate=500000,
                 receive_own_messages=False) as b:

        b.send(can.Message(arbitration_id=0, data=[0x01]))

        messages = []

        msg = b.recv(timeout=initial_timeout)
        while msg is not None:
            messages.append(msg)
            msg = b.recv(timeout=3)

        leds_id = [msg.arbitration_id for msg in messages]

    return leds_id
```
### Additional context

zariiii9003, 2022-05-04:
Something like [this](https://github.com/hardbyte/python-can/blob/796b52586a5d891e4e2077b423de40cba07f8b44/can/interface.py#L125)?

GiorgioSgl, 2022-05-04:
This

> Something like [this](https://github.com/hardbyte/python-can/blob/796b52586a5d891e4e2077b423de40cba07f8b44/can/interface.py#L125)?

This seems more to identify the interface available, and not as in my case to identify the device available in my can-network.

GiorgioSgl, 2022-05-05:
> That is dependent on the protocol used on the bus. You have to do different logic depending on if it J1939, UDS, CANopen et.c. This library does not contain any logic for specific protocols so either you need to figure out how to do it or use a library specific for your protocol.

That's help! So actually it's depending on the protcol and with the library is not possible. I'm going to close it. Thanks!