hardbyte/python-can #1305: CAN FD - serial line?

jagauthier, 2022-05-04:
I'm sorry to open a "bug" report for a support issue.
I've been reading through the documentation, and I know that CAN FD support is present.
But I can't seem to find out of it's supported via serial over CAN (slcan)

I know this is an ASCII protocol for traditional CAN: 
http://www.can232.com/docs/can232_v3.pdf

But I am having a hard time finding definitive information about this protocol and CAN FD.

jagauthier, 2022-05-04:
Thanks for your elaborate response!  I actually have custom built hardware and I am trying to figure out the best method to use for both Windows and Linux functionality.  Fortunately, this will not be used for high speed communications. Mostly status and commands, but I am using CAN FD because the network will absolutely support higher speeds.  candlelight firmware with binary USB seems promising, but I am not very familiar with gs_usb at this point either.

hartkopp, 2022-05-04:
The candlelight firmware implements the GS_USB protocol which either works on Linux/SocketCAN out of the box and obviously also directly which python-can:

https://github.com/hardbyte/python-can/blob/develop/can/interfaces/gs_usb.py

The CAN FD support is pretty new (in the firmware and in the Linux kernel). But IMHO this could be the most promising way for your custom built hardware to follow.

jagauthier, 2022-05-04:
It also looks like the firmware "also implements WCID USB descriptors" so it can be used on Windows.
I'm not sure if I can get away with Python-can, which I already use, but I think I can use some other gs-usb based modules.

closed 2022-05-10 by jagauthier

thestumbler, 2022-09-19:
I would like to second this request. I have three different models serial-CAN devices that support FD: Grid Connect CAN232 FD, Canable 2.0, and Mjbots FdCanUsb.