hardbyte/python-can #1310: Event type message cannot display in bus

catmemo, 2022-05-07:
### Describe the bug
I tried to trigger an event type message but I did not find it in bus.

### To Reproduce
Run below code and trigger an event type message:
```
import can

def send():

    bus = can.interface.Bus(bustype='vector', app_name=None, channel=0, bitrate=500000)
    for msg in bus:
        print(msg)
        print('\n')

if __name__ == "__main__":
    send()
```

### Expected behavior
The event type message can be printed in bus.

### Additional context

OS and version: win10
Python version:3.7.2
python-can version: latest
python-can interface/s (if applicable): No

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

zariiii9003, 2022-05-07:
What kind of message do you expect? If you do not set `app_name` or `serial` then you connect to the Virtual Channel 1. Is any application sending messages on Virtual Channel 1?

catmemo, 2022-05-09:
> What kind of message do you expect? If you do not set `app_name` or `serial` then you connect to the Virtual Channel 1. Is any application sending messages on Virtual Channel 1?

Event type message means I will send only 3 frames of this message. And I connect with CANalyzer.

catmemo, 2022-05-10:
Sorry, I made a mistake, it's OK to get all messages.

closed 2022-05-10 by catmemo