hardbyte/python-can #1312: Notifier bug under Windows

gribera, 2022-05-15:
### Describe the bug

When I try to instantiate Notifier using robotell interface it fails on Windows. In Linux it works without problems

### To Reproduce

```python
import can

bus = can.interface.Bus(bustype='robotell',
                             channel='COM5',
                             ttyBaudrate=115200,
                             bitrate=250000)
                             
buffer = can.BufferedReader()
notifier = can.Notifier(bus, [buffer])
```

```
Traceback (most recent call last):
  File "C:\Users\locaa\Documents\Microredes\test2.py", line 10, in <module>
    notifier = can.Notifier(bus, [can_listener])
  File "C:\Users\locaa\AppData\Local\Programs\Python\Python310\lib\site-packages\can\notifier.py", line 59, in __init__
    self.add_bus(each_bus)
  File "C:\Users\locaa\AppData\Local\Programs\Python\Python310\lib\site-packages\can\notifier.py", line 69, in add_bus
    reader = bus.fileno()
  File "C:\Users\locaa\AppData\Local\Programs\Python\Python310\lib\site-packages\can\interfaces\robotell.py", line 371, in fileno
    return self.serialPortOrig.fileno()
io.UnsupportedOperation: fileno
```

### Expected behavior

Notifier should be instantiated without any error.

### Additional context

OS and version: Windows 10
Python version: Python 3.10.4
python-can version: 4.0.0

zariiii9003, 2022-05-19:
The `fileno()` method of the robotell interface should call `super().fileno()` on windows. This raises a NotImplementedError which is expected and handled by the notifier.

Would you like to create a pull request?

gribera, 2022-05-20:
Thanks for your reply @zariiii9003, I sent a pull request with a fix that works. It is based on the `fileno()` method of slcan BUS.