hardbyte/python-can #1315: Logged .asc file can't open in CANoe 14

tienlocnguyen, 2022-05-23:
### Describe the bug
Logged .asc file by python-can is not able to open in CANoe 14

### To Reproduce
- Record .asc file by python-can
- Try to replay logged file by CANoe 14

### Expected behavior
Can replay log by CANoe 14

### Additional context

OS and version: Window 10
Python version: 3.8
python-can version: 4.0.0(https://github.com/hardbyte/python-can/releases/download/4.0.0/python_can-4.0.0-py3-none-any.whl)
python-can interface/s (if applicable): 

### Root cause
Date information in header of .asc file contain 6 digits of milisecond and somehow violate CANoe 14 rule. 3 digits is accepted
Not work: date Mon May 23 03:29:03.103**xxx** PM 2022
Worked: date Mon May 23 03:29:03.103 PM 2022

### Propose fix
Use same logic to get datetime when write begin trigger block
File [asc.py](https://github.com/hardbyte/python-can/blob/develop/can/io/asc.py), line 368-369
```python
        now = datetime.now().strftime(self.FORMAT_START_OF_FILE_DATE)
        self.file.write(f"date {now}\n")
```
```python
        now = datetime.now().timestamp()
        mlsec = repr(now).split(".")[1][:-3]
        formatted_date = time.strftime(
            self.FORMAT_DATE.format(mlsec), time.localtime(now)
        )
        self.file.write(f"date {formatted_date}\n")
```

zariiii9003, 2022-06-10:
Would you like to create a PR?

Tian-Jionglu, 2022-08-04:
Thanks to @tienlocnguyen , this solution solved my issue.
I created a PR #1357  to fix this issue. Sightly difference, since `can_logconvert` should also be fixed.

PR #1362 (2022-08-11): Write 3 ms digits to ascii to resolve CANoe bug
- The ascii writer will now only write 3 micro second decimal places to a log file. 
- The ascii format currently is not setup to append. I recenly added in **options to the logger script. **options are only required for
the rolling logger.

closes #1315
closes #1357
closed 2022-08-11 by zariiii9003