hardbyte/python-can #1319: Reflecting xldriver exception error code

Allabenarush, 2022-05-26:
When trying to send a large amount of CAN messages to Vector I get an exception from xldriver:
can.interfaces.vector.exceptions.VectorOperationError: xlCanTransmitEx failed (XL_ERR_QUEUE_IS_FULL) [Error Code 11]

I seem to have a problem catching this exception because its raised in the check_status_operation method in xldriver.py

Here is the trace:
xlCanTransmitEx failed (XL_ERR_QUEUE_IS_FULL) [Error Code 11]
Traceback (most recent call last):
  File "C:\Users\labuser\PycharmProjects\gk_auto_infra\auto_infra\generators\can_gen\can_descriptor_manager.py", line 222, in _send_vector_can_msg_list
    tx_bus.send(msg, 60)
  File "C:\Users\labuser\PycharmProjects\gk_auto_infra\venv\lib\site-packages\can\interfaces\vector\canlib.py", line 536, in send
    self._send_sequence([msg])
  File "C:\Users\labuser\PycharmProjects\gk_auto_infra\venv\lib\site-packages\can\interfaces\vector\canlib.py", line 541, in _send_sequence
    return self._send_can_fd_msg_sequence(msgs)
  File "C:\Users\labuser\PycharmProjects\gk_auto_infra\venv\lib\site-packages\can\interfaces\vector\canlib.py", line 594, in _send_can_fd_msg_sequence
    self.xldriver.xlCanTransmitEx(
  File "C:\Users\labuser\PycharmProjects\gk_auto_infra\venv\lib\site-packages\can\interfaces\vector\xldriver.py", line 37, in check_status_operation
    raise VectorOperationError(
can.interfaces.vector.exceptions.VectorOperationError: xlCanTransmitEx failed (XL_ERR_QUEUE_IS_FULL) [Error Code 11]
xlCanTransmitEx failed (XL_ERR_QUEUE_IS_FULL) [Error Code 11]

Is it possible to receive the error code in the send method in C:\Users\labuser\PycharmProjects\gk_auto_infra\venv\lib\site-packages\can\interfaces\vector\canlib.py?

Thanks in advance

zariiii9003, 2022-05-26:
Each python-can exception has an `error_code` attribute
https://github.com/hardbyte/python-can/blob/4cb2f2f6fbf6bd378a4e804b5c737a6128caea1d/can/exceptions.py#L57

zariiii9003, 2022-06-10:
I assume this is resolved. Feel free to reopen if you have further questions.

closed 2022-06-10 by zariiii9003