hardbyte/python-can #1320: Add documentation for how to build the docs to README

j-c-cook, 2022-05-27:
### Is your feature request related to a problem? Please describe.

Contributors often need to edit documentation. It is nice to be able to build the documentation locally prior to pushing up changes to make sure everything looks okay. I don't currently see any documentation for how to build the docs. 

### Describe the solution you'd like

A new section in the README dedicated to building the docs.

zariiii9003, 2022-05-27:
There's a [section in the docs](https://python-can.readthedocs.io/en/master/development.html#building-installing) for it.

j-c-cook, 2022-05-28:
I hadn't noticed that. I should have looked into the contributing docs closer.

closed 2022-05-28 by j-c-cook