hardbyte/python-can #1321: linux canutils cangw

xeno27, 2022-06-01:
### Is your feature request related to a problem? Please describe.
It would be nice to have an simular api to cangw like the canutils cangw, iam working a lot with the internal cangw function, and it is actualy great ! but iam not sure  if the cangw things are here at the right position..

### Describe the solution you'd like
It would be great to have an interface to add cangw functions and remove them. Here are a lot of things possible (changing msg contend by passing the gateway, changing the checksum and and and) and all this things are running in the Linux kernel. So it is incredibly fast ! 

### Describe alternatives you've considered
just use the c interface. This works for me, but it gives a lot of freedom to use this things in python can

### Additional context
This is first a general discussion about this. 
If i have the time i would start to implement this things (with help from your side?) 

For more infos about the cangw : https://github.com/linux-can/can-utils

hartkopp, 2022-06-06:
Hi @xeno27 ,
the can-gw is a Linux-only feature which makes use of PF_NETLINK sockets to configure the can-gw routing jobs with netlink messages.

That means the can-gw is more a PF_NETLINK feature than a PF_CAN feature. I would suggest to do some investigations how PF_NETLINK can be used from Python and maybe create your own library?

xeno27, 2022-06-15:
Hi @hartkopp ,

I think this will be the best way.

@jayceslesar i think an own lib make sense.

closed 2022-06-15 by xeno27