hardbyte/python-can #1324: send_periodic() only sends messages once

melav99, 2022-06-03:
I still have the same issue as  #292 even with the newest release. However, once I add a time.sleep(any value) it works for an infinite time. 

```python
  import cantools
  import can
  import time
  from pprint import pprint 
  
  bus = can.interface.Bus(channel='can0', bustype='socketcan', bitrate='500000')
  db = cantools.database.load_file('CAN_v2.2.kcd')

  LLC_Voltage_Limits_data = db.encode_message('LLC_Voltage_Limits_3', {
        'Input_voltage_min_3': 0, 'Input_voltage_max_3': 750, 'Output_voltage_min_3': 0, 'Output_voltage_max_3': 750
        }
    )
    msg = can.Message(
            arbitration_id=0x30047,
            data=LLC_Voltage_Limits_data,
            is_extended_id=True,
    )  

    task = bus.send_periodic(msg, 0.1)
    assert isinstance(task, can.CyclicSendTaskABC)
```

christiansandberg, 2022-06-08:
The issue is that your script exits immediately. Add a time.sleep() or wait for some input from the user.

melav99, 2022-07-25:
I added it to a loop with conditions and it works now. Thank you!

closed 2022-07-25 by melav99