hardbyte/python-can #1325: AttributeError: 'list' object has no attribute 'channel'

melav99, 2022-06-03:
When running the exact code in the [cyclic.py example](https://github.com/hardbyte/python-can/blob/develop/examples/cyclic.py), I get the error: AttributeError: 'list' object has no attribute 'channel'
 
I am using the latest versions of python and the packages. 

Addition to error message:

``` 
Traceback (most recent call last):
  File "/root/no.py", line 139, in <module>
    cyclic_multiple_send(BUS)
  File "/root/no.py", line 63, in cyclic_multiple_send
    task = bus.send_periodic(messages, 1)
  File "/usr/lib/python3.9/site-packages/can/bus.py", line 208, in send_periodic
  File "/usr/lib/python3.9/site-packages/can/interfaces/socketcan/socketcan.py", line 710, in _send_periodic_internal
```

zariiii9003, 2022-06-03:
I am pretty sure that your are not using the latest version. Probably 3.3.4 or older. Upgrade to 4.0

melav99, 2022-06-07:
You are correct! Sorry about that, thank you!

closed 2022-06-07 by melav99