hardbyte/python-can #1326: Enhance the `can.logger` to check to see if the log file exists and append if there rather than overwrite

j-c-cook, 2022-06-03:
### Is your feature request related to a problem? Please describe.

Consider `can.logger` being used on an embedded Linux system. A `python can.logger ...` process is started as a daemon (`/etc/init.d/`) in boot and killed in shutdown. When the system is restarted, the existing log file on the device and its contents is currently overwritten. This results in a loss of data. 

### Describe the solution you'd like

Upon executing `python can.logger ...` the program should check to see if the specified log file exists. If it does exist, then the file should be appended to. 

### Describe alternatives you've considered

An optional argument could be added (perhaps `-a` or `-w`) to give the user the option as to whether an existing file, if found, should be overwritten or appended to. 

### Additional context

P.S. Feel free to assign this issue to me. I should be around to it by Sunday or Monday.

closed 2022-06-10 by zariiii9003

j-c-cook, 2022-06-10:
@zariiii9003 Given the merge of the new feature provided in #1327, could you please consider the next release to be `v4.1.0` rather than `v4.0.1`? If we go by https://semver.org/ for release versions, then the next release would now be considered a MINOR release. 

> Given a version number MAJOR.MINOR.PATCH, increment the:
> 
> 1. MAJOR version when you make incompatible API changes,
> 2. MINOR version when you add functionality in a backwards compatible manner, and
> 3. PATCH version when you make backwards compatible bug fixes.

I think the easiest way to do this would be to relabel [Milestone 4.1.0](https://github.com/hardbyte/python-can/milestone/10) to `Milestone 4.2.0`. Then relabel [Milestone 4.0.1](https://github.com/hardbyte/python-can/milestone/11) to `Milestone 4.1`. Then any issues that should be included sooner rather than later can be bumped from `Milestone 4.2` to `Milestone 4.1`.

zariiii9003, 2022-06-10:
I don't mind, but i cannot create releases anyway. I will ask @hardbyte once the other bugfixes are reviewed and merged.

PR #1327 (2022-06-10): Enhance `can.logger` to consider the `append` option
The goal here appears to be to give access to the boolean `append` option of `Printer`. 

https://github.com/hardbyte/python-can/blob/0d7f65f74b8b015fb650e9910a920323023f310d/can/io/printer.py#L16-L40

closes #1326