hardbyte/python-can #1330: Could not import vxlapi: Could not find vxlapi64

jp213, 2022-06-10:
Hello, I seem to be having an issue when I try to run a device simulation I made. I installed the drivers for python-can and ensured that vxlapi.py is located in the same library as canlib.py. The error is listed below. Thank you!

Could not import vxlapi: Could not find module 'vxlapi64' (or one of its dependencies). Try using the full path with constructor syntax.
Traceback (most recent call last):
  File "c:\Users\U294888\AppData\Local\Programs\Python\Python310\jlg_utils-develop@c05b84cf3c6\J1939\src\device_simulation\Single_tilt_sensor.py", line 115, in <module>
    main()
  File "c:\Users\U294888\AppData\Local\Programs\Python\Python310\jlg_utils-develop@c05b84cf3c6\J1939\src\device_simulation\Single_tilt_sensor.py", line 104, in main
    sensor = SimTiltSensor(source_address=0xC0)
  File "c:\Users\U294888\AppData\Local\Programs\Python\Python310\jlg_utils-develop@c05b84cf3c6\J1939\src\device_simulation\Single_tilt_sensor.py", line 29, in __init__
    self.device = simulated_device.SimulatedDevice(
  File "c:\Users\U294888\AppData\Local\Programs\Python\Python310\jlg_utils-develop@c05b84cf3c6\J1939\src\device_simulation\simulated_device.py", line 39, in __init__
    self._comms = shared.comms_setup.CommSetup(
  File "c:\Users\U294888\AppData\Local\Programs\Python\Python310\jlg_utils-develop@c05b84cf3c6\J1939\src\device_simulation\shared\comms_setup.py", line 29, in __init__
    self.connect()
  File "c:\Users\U294888\AppData\Local\Programs\Python\Python310\jlg_utils-develop@c05b84cf3c6\J1939\src\device_simulation\shared\comms_setup.py", line 64, in connect
    self._ecu.connect(
  File "C:\Users\U294888\AppData\Local\Programs\Python\Python310\lib\site-packages\j1939\electronic_control_unit.py", line 119, in connect
    self._bus = can.interface.Bus(*args, **kwargs)
  File "C:\Users\U294888\AppData\Local\Programs\Python\Python310\lib\site-packages\can\interface.py", line 127, in __new__
    return cls(channel, *args, **kwargs)
  File "C:\Users\U294888\AppData\Local\Programs\Python\Python310\lib\site-packages\can\interfaces\vector\canlib.py", line 81, in __init__
    raise ImportError("The Vector API has not been loaded")
ImportError: The Vector API has not been loaded

zariiii9003, 2022-06-10:
There's a template for bug reports. Please take a look and update your post.

closed 2022-06-10 by jp213

SandeepMVS, 2024-02-16:
ima trying to use python-can with send_one(), but getting below error. can any one please help in fix same.
Could not import vxlapi: Could not find module 'vxlapi64' (or one of its dependencies). Try using the full path with constructor syntax.
Traceback (most recent call last):
  File "D:\Python programs\new.py", line 20, in <module>
    send_one()
  File "D:\Python programs\new.py", line 9, in send_one
    with can.Bus(interface='vector', app_name='CANoe', channel=1,bitrate=250000) as bus:
  File "C:\Users\ms2\AppData\Roaming\Python\Python310\site-packages\can\util.py", line 378, in wrapper
    return f(*args, **kwargs)
  File "C:\Users\ms2\AppData\Roaming\Python\Python310\site-packages\can\interface.py", line 135, in Bus
    bus = cls(channel, **kwargs)
  File "C:\Users\ms2\AppData\Roaming\Python\Python310\site-packages\can\util.py", line 378, in wrapper
    return f(*args, **kwargs)
  File "C:\Users\ms2\AppData\Roaming\Python\Python310\site-packages\can\interfaces\vector\canlib.py", line 171, in __init__
    raise CanInterfaceNotImplementedError("The Vector API has not been loaded")
can.exceptions.CanInterfaceNotImplementedError: The Vector API has not been loaded