hardbyte/python-can #1331: CAN-FD initialization error on PEAK device

sortkim, 2022-06-14:
### Describe the bug
CAN-FD initialization error on PEAK device (PCANUSB-FD)
vector can works well with CAN-FD.

### To Reproduce
    h_can = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000, fd=True) # fail

### Expected behavior
return bus object address

### Additional context

OS and version: Windows10
Python version : 3.9.12
python-can version : 4.0.0
python-can interface/s (if applicable): PCAN-USB FD

<details><summary>Traceback and logs</summary>
none

none

import can

try:
    # h_can = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000) # pass
    h_can = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000, fd=True) # fail
    # h_can = can.interface.Bus(bustype='vector', channel=2, bitrate=500000, fd=True) # pass
except:
    h_can = None

if h_can is not None:
    tx_msg = can.Message(arbitration_id=100, dlc= 12, data=[0, 0, 0, 0, 0, 0, 0, 0, 1, 2, 3, 4], is_fd=True)
    # tx_msg = can.Message(arbitration_id=100, dlc= 8, data=[0, 0, 0, 0, 0, 0, 0, 0])
    h_can.send(tx_msg)

</details>

zariiii9003, 2022-06-14:
Traceback?

skipper85, 2022-06-17:
Hi there,

I also use PEAK system IPEH-004002 USBCAN converter with CAN FD support in combination with python-can, but also get error during initialization.

### Context
OS and version: Windows 10
Python version: 3.9.13
python-can version: 4.0.0
python-can interface/s (if applicable): PCAN-USB FD

### Code
`bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000, fd=True)`

### Traceback
```
runfile('C:/Users/a1084371/python_can_ni/test_peak_pcan.py', wdir='C:/Users/a1084371/python_can_ni')
Traceback (most recent call last):

  File "C:\Users\a1084371\python_can_ni\ni_env_3_9\lib\site-packages\spyder_kernels\py3compat.py", line 356, in compat_exec
    exec(code, globals, locals)

  File "c:\users\a1084371\python_can_ni\test_peak_pcan.py", line 59, in <module>
    test_play_blf()

  File "c:\users\a1084371\python_can_ni\test_peak_pcan.py", line 29, in test_play_blf
    bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000, fd=True)

  File "C:\Users\a1084371\python_can_ni\ni_env_3_9\lib\site-packages\can\interface.py", line 120, in __new__
    bus = cls(channel, *args, **kwargs)

  File "C:\Users\a1084371\python_can_ni\ni_env_3_9\lib\site-packages\can\interfaces\pcan\pcan.py", line 246, in __init__
    raise PcanCanInitializationError(self._get_formatted_error(result))

PcanCanInitializationError: An unknown error has occurred
```

During debug the following variables in PcanBus __init__ were noticed:
result: 65536
self.fd_bitrate: b'f_clock_mhz=None'
self.m_PcanHandle: c_ushort(81)

I would expect a different value for the fd_bitrate.

skipper85, 2022-06-17:
Hi there,

Apparently the setup for the CAN-FD is quite different from normal CAN.
When using the following initialization

```
bus = can.interface.Bus(
        bustype='pcan',
        channel='PCAN_USBBUS1',
        fd=True,
        f_clock_mhz=80,
        nom_brp=1,
        nom_tseg1=138,
        nom_tseg2=21,
        nom_sjw=21,
        data_brp=1,
        data_tseg1=80,
        data_tseg2=79,
        data_sjw=79)
```

the device works fine.
I used [Kvaser calculator](https://www.kvaser.com/support/calculators/can-fd-bit-timing-calculator/) to calculate all the CAN-FD settings.

sortkim, 2022-06-20:
Thank you everyone.

It works fine now.
I took your advice into account.

j-c-cook, 2022-08-06:
I have a [PCAN-USB](https://www.peak-system.com/PCAN-USB.199.0.html?&L=1), but not the [PCAN-USB FD](https://www.peak-system.com/PCAN-USB-FD.365.0.html?&L=1), so I don't think I have the right hardware to help with this (if I'm wrong, let me know). Unfortunately, the FD device is out of stock. I'll ask around and see if I can locate a PEAK FD device to borrow for long enough to resolve this issue.