hardbyte/python-can #1334: Error importing IXXATBus on Linux

MartinALL91, 2022-06-21:
### Describe the bug

Error when importing IXXATBus.

### Traceback

```
from can.interfaces.ixxat.canlib import IXXATBus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/sw-linux/Documents/GitProjects/ingenialink-python/ingenialink-py/lib/python3.8/site-packages/can/interfaces/ixxat/__init__.py", line 9, in <module>
    from can.interfaces.ixxat.canlib import IXXATBus
  File "/home/sw-linux/Documents/GitProjects/ingenialink-python/ingenialink-py/lib/python3.8/site-packages/can/interfaces/ixxat/canlib.py", line 26, in <module>
    from can.ctypesutil import CLibrary, HANDLE, PHANDLE, HRESULT as ctypes_HRESULT
ImportError: cannot import name 'HRESULT' from 'can.ctypesutil' (/home/sw-linux/Documents/GitProjects/ingenialink-python/ingenialink-py/lib/python3.8/site-packages/can/ctypesutil.py)

```

### Additional context

OS and version: Ubuntu 20.04
Python version: 3.8
python-can version: 3.3.4
python-can interface/s (if applicable):

zariiii9003, 2022-06-21:
The [docs ](https://python-can.readthedocs.io/en/master/interfaces/ixxat.html)say:

> The Linux ECI SDK is currently unsupported, however on Linux some devices are supported with [SocketCAN](https://python-can.readthedocs.io/en/master/interfaces/socketcan.html).

MartinALL91, 2022-06-23:
Ok, thanks. I'll close the issue.

closed 2022-06-23 by MartinALL91