hardbyte/python-can #1350: Hi Everyone,

Oemm868, 2022-07-26:
### Is your feature request related to a problem? Please describe.

### Describe the solution you'd like

### Describe alternatives you've considered

### Additional context

j-c-cook, 2022-08-06:
Could you provide more information? What object are you referring to? Do you have a minimal script that I could get started with?

wallem89, 2023-06-20:
The receive own messages function does work with a SLCAN, at least with a VSCOM adapter that I am using. I expect the problem might be in the listener. Did you try to receive the message without the notifier?

So try something like this:
```
for msg in can_bus:
    print(f"Message ID {msg.arbitration_id} received with data {msg.data}")
```

Oemm868, 2023-07-06:
Hi wallem89, I have checked your procedure but had no success. I am checking if it is a hardware problem with my adapter. It seems likely that this problem is caused there. I have a great ambition to solve the problem, but the priority of my project is currently low. I close this issue first, in case it is not a "My Hardware Problem" I open the issue again.

closed 2023-07-06 by Oemm868