hardbyte/python-can #1352: USB2CAN

jkljkjkjk, 2022-07-29:
Hi there

I am new here and I am wondering if the lib supports ZLG USBCAN series products?

I read the docs and did not see the examples of usb2can relevant, is it really supported, or any examples there?

thank you very much

j-c-cook, 2022-08-06:
Have you ordered the product and tested it? [SocketCan](https://docs.kernel.org/networking/can.html) in the Linux kernel is network based, so the first step after plugging in the device would be to [bring up the CAN interface](https://elinux.org/Bringing_CAN_interface_up). Then initialize your `Bus` with the "socketcan" `channel`. 

I use a PCAN in Linux in a similar fashion described above. No additional drivers required. See `python-can` docs [here](: https://python-can.readthedocs.io/en/develop/interfaces/socketcan.html).

jkljkjkjk, 2022-08-08:
Yes, I've ordered the product, but still on the way. I understand that SocketCan is available on Linux and it is network based, however as far as I know, ZLG USBCAN does not support socketcan, that's why I was wondering if python-can supports ZLG usbcan series product, if it does, how to announce bus type as well as channel name.

PCAN no matter PCIE or USB types both support socketcan, I used PCIE PCAN before. But to be honest that is little bit expensive comparing to ZLG

thanks a lot