hardbyte/python-can #1353: Initialize Vector Dongle VN1610 with CAN-FD Fails

skipper85, 2022-07-29:
### Describe the bug
When I try to initialize the can.bus() for the VN1610 vector dongle with the fd parameter I get the following error.
`
can.interfaces.vector.exceptions.VectorInitializationError: xlCanFdSetConfiguration failed (XL_ERR_WRONG_PARAMETER) [Error Code 101]
`

Without the fd parameters the bus initialize, but for normal CAN.

### To Reproduce
Initialize the bus with the fd parameter.

### Expected behavior
Expect the CAN bus initialized with CAN-FD for the vector dongle.

### Additional context
No issues with the peak-systems PCAN dongle.

OS and version: Windows 10
Python version: 3.9.13
python-can version: python-can==4.0.0
python-can interface/s (if applicable): vector VN1610 dongle, driver version 20.30.14

`
Traceback (most recent call last):
  File "c:\users\a1084371\onedrive - prod - sensata.com azure ad (sso.sensata.com) (not o365)\documents\hengelo-qlab\dccb_daimler\tests\test_vector.py", line 89, in <module>
    canbus = can.Bus(
  File "C:\Users\a1084371\OneDrive - PROD - sensata.com Azure AD (sso.sensata.com) (Not O365)\Documents\hengelo-qlab\dccb_daimler\ni_env_3_9\lib\site-packages\can\interface.py", line 120, in __new__
    bus = cls(channel, *args, **kwargs)
  File "C:\Users\a1084371\OneDrive - PROD - sensata.com Azure AD (sso.sensata.com) (Not O365)\Documents\hengelo-qlab\dccb_daimler\ni_env_3_9\lib\site-packages\can\util.py", line 320, in wrapper
    return f(*args, **kwargs)
  File "C:\Users\a1084371\OneDrive - PROD - sensata.com Azure AD (sso.sensata.com) (Not O365)\Documents\hengelo-qlab\dccb_daimler\ni_env_3_9\lib\site-packages\can\interfaces\vector\canlib.py", line 279, in __init__
    self.xldriver.xlCanFdSetConfiguration(
  File "C:\Users\a1084371\OneDrive - PROD - sensata.com Azure AD (sso.sensata.com) (Not O365)\Documents\hengelo-qlab\dccb_daimler\ni_env_3_9\lib\site-packages\can\interfaces\vector\xldriver.py", line 46, in check_status_initialization
    raise VectorInitializationError(
can.interfaces.vector.exceptions.VectorInitializationError: xlCanFdSetConfiguration failed (XL_ERR_WRONG_PARAMETER) [Error Code 101]
`

``` python
canbus = can.Bus(
            channel=1,
            bustype="vector",
            fd=True, # not set anymore in the xldriver?
            bitrate=500000,
            data_bitrate=5000000,
            sjw_abr=40,
            tseg1_abr=127,
            tseg2_abr=32,
            sjw_dbr=4,
            tseg1_dbr=9,
            tseg2_dbr=6,
        )
```
</details>

zariiii9003, 2022-07-31:
I'm not sure about this but your arbitration sync jump width seems implausible to me. It cannot be greater than tseg1_abr or tseg2_abr.

j-c-cook, 2022-08-06:
@skipper85 Were you able to get this functioning based on the comment from @zariiii9003? Is it true that `sjw_abr` must be less than `tseg1_abr` and `tsg2_abr`?

skipper85, 2022-08-08:
Hi, currently the dongle is in use, I hope I can use it this week and do some debugging. When available I will try @zariiii9003 suggestion with the settings and in the meanwhile I will dig in about those settings.

skipper85, 2022-08-10:
Hi,

I tried again with the vector dongle VN1610
When I use the following code:
``` Python
canbus = can.Bus(
        channel=1,
        bustype="vector",
        app_name="python_can",
        fd=True,
        bitrate=500000,
        data_bitrate=5000000
    )
```
I still receive the following exception:
`
can.interfaces.vector.exceptions.VectorInitializationError: xlCanFdSetConfiguration failed (XL_ERR_WRONG_PARAMETER) [Error Code 101]
`

zariiii9003, 2022-08-10:
You need to set the correct bit timings for your bus. We cannot tell you how your bus is configured. If you initialize the bus with valid parameters then it works fine.

```python
import can

bus = can.Bus(
    channel=1,
    bustype="vector",
    fd=True,
    bitrate=500_000,
    data_bitrate=5_000_000,
    sjw_abr=2,
    tseg1_abr=11,
    tseg2_abr=4,
    sjw_dbr=2,
    tseg1_dbr=5,
    tseg2_dbr=2,
)
```

This should should initialize the bus with no exception. But you will get error messages if the settings are wrong for your application.

skipper85, 2022-08-10:
Hi, thanks for your help, it does work now, also with my own settings I recalculated for the bit timings.
```
'channel':1,
'bustype':"vector",
'fd':True,
'bitrate':500_000,
'data_bitrate':5_000_000,
'sjw_abr':32,
'tseg1_abr':127,
'tseg2_abr':32,
'sjw_dbr':6,
'tseg1_dbr':9,
'tseg2_dbr':6
```

closed 2022-09-11 by zariiii9003