hardbyte/python-can #1356: Timestamp error on python

AbdulkadirEroglu-AptiRobotics, 2022-08-03:
### Describe the bug

When sending data i am getting struct error.
İt seems like when packing timestamp there is a multiplication step and that gets data to over limit. 

### To Reproduce

send some data with time stamp field filled with python time modules time.time() function.

### Expected behavior

data to be send through serial interface

### Additional context

OS and version: Ubuntu 22.04 - Ubuntu 20.04
Python version: 3.10, 3.9
python-can version: 4.0.0
python-can interface/s (if applicable): virtual can with socat tool

[image]

<details><summary>Traceback and logs</summary>

[image]

j-c-cook, 2022-08-06:
What is the script `tests/SanalCan.py`? Could you please upload a minimal script that triggers the bug you are describing? I am unable to follow your "To Reproduce" section well enough to reproduce your script.

AbdulkadirEroglu-AptiRobotics, 2022-08-08:
> What is the script tests/SanalCan.py? 

It is a script that sends random data to can network.
Here is a full code to repreduce the error

`import can, time`
`bus = can.interface.Bus(interface="serial", channel="/dev/ttyS11")`
`msg = can.Message(time.time(), 1, False, data=[0x11, 0x22, 0x33], is_rx=False)`
`bus.send(msg)`

zariiii9003, 2022-08-08:
You can't use the serial bus with `time.time()`. The documentation says [Usually s, ms or µs since start of the device](https://python-can.readthedocs.io/en/master/interfaces/serial.html#serial-frame-format)

AbdulkadirEroglu-81, 2022-08-09:
Oh, I didn't see that. I am closing the issue.

closed 2022-08-09 by AbdulkadirEroglu-81