hardbyte/python-can #1368: Vector app name?  Trying to read two channels with different baud rates

CathySmith17, 2022-08-08:
Hi,

I am trying to read 2 channels at once with my Vector tool 1610.  The two channels have different baud rates.

I tried the following formatting but that throws an error
can_bus = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=[0, 1], bitrate=[500000, 125000])
=> TypeError: int() argument must be a string, a bytes-like object or a number, not 'set'

I also tried creating a custom Canalyzer .cfg and then tried to load it as follows

can_bus = can.interface.Bus(bustype='vector', app_name='CustomCfg.cfg', channel=[0, 1])
=> can.interfaces.vector.exceptions.VectorInitializationError: xlGetApplConfig failed (XL_ERR_WRONG_PARAMETER) [Error Code 101]
this throws another error. What kind of file does the app_name want as input. and where should it be stored?

Regards, 
Cathy

zariiii9003, 2022-08-08:
app_name refers to the application name that you defined in Vector Hardware Config.

You cannot set multiple bitrates directly, the python-can API does not support that. But you can use the Vector XL API directly to set the channel bitrates:

```python3
from can.interfaces.vector import VectorBus

if __name__ == '__main__':
    bus = VectorBus(
        serial=100,
        channel=[0, 1],
        bitrate=500_000,
    )
    bus.xldriver.xlCanSetChannelBitrate(bus.port_handle, bus.channel_masks[0], 500_000)
    bus.xldriver.xlCanSetChannelBitrate(bus.port_handle, bus.channel_masks[1], 125_000)

```

CathySmith17, 2022-08-09:
Thanks for replying, understand the app part now!

Unfortunately using the Vector API doesn't work either:

Either it runs through but doesnt actually read any bus at all or it gives one of two errors

in check_status_initialization
    result, xlGetErrorString(result).decode(), function.__name__
can.interfaces.vector.exceptions.VectorInitializationError: xlCanSetChannelBitrate failed (XL_ERR_INVALID_ACCESS) [Error Code 112]

if i replace serial=100 by app_name='CANAlyzer':
in check_status_initialization
    result, xlGetErrorString(result).decode(), function.__name__
can.interfaces.vector.exceptions.VectorInitializationError: xlCanSetChannelBitrate failed (XL_ERROR) [Error Code 255]

zariiii9003, 2022-08-09:
You need init access, so you need to be the first application on the bus. Otherwise you don't have the right to change the bus settings. Look for the `Init` flag into your Vector HW config:
[image]

CathySmith17, 2022-08-11:
Hey zariiii9003,

I wanted to thank you very much taking the time to help me and pointing me in the right direction. Unfortunately the command still doesn't work. The status line you marked in the screenshot never shows up when using xlCanSetChannelBitrate. It does however when using: can_bus2 = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=[1], bitrate=125000)

For anybody having the same issue i managed to get it to work using either one of the following 2 methods
1) In screenshot above , left from "Default CAN baud rate" one can simple set default channel baud rate, then there is no need to do it programmatically.  =>
         can_bus = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=[0, 1])

2) Set the bit rate  of each channel individually and then create a new bus
         can_bus1 = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=[0], bitrate=500000)
         can_bus2 = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=[1], bitrate=125000)
         can_bus = VectorBus(
             serial=XXXX,    # This needs to be your actual serial which shows up in the blacked out part in the image above
            channel=[0, 1],
         )
        del can_bus1, can_bus2

zariiii9003, 2022-11-14:
While there is still no official API for that, you could use private methods of the `VectorBus` with the latest develop branch. 

```python3
from can.interfaces.vector import VectorBus

bus = VectorBus(channel=[0, 1], serial=100, bitrate=500_000)
bus._set_bitrate_can(channel=0, bitrate=500_000)
bus._set_bitrate_can(channel=1, bitrate=125_000)
```

Of course private methods might change in the future, but maybe someone needs this. I will close this issue now.

closed 2022-11-14 by zariiii9003