hardbyte/python-can #1372: The actual playback time is much larger than the timestamp

zixuan-su, 2022-08-12:
### Describe the bug

I have a 120 s long message ，then  I find  that It takes me 324.2 s to play. The Timestamp is still about 120 s, but  the difference from the actual playback time is nearly 2.7 times .    The file is https://github.com/hardbyte/python-can/blob/develop/can/player.py 

### To Reproduce

### Expected behavior

### Additional context

OS and version: Ubuntu 1804
Python version: 3.9.9
python-can version: 4.0.0
python-can interface/s (if applicable):

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

zariiii9003, 2022-08-12:
Could you replace this 
https://github.com/hardbyte/python-can/blob/3a7c80eb24d253b092b038f45be9d36942c2ab2d/can/io/player.py#L154

with this

```python3
            if sleep_period > 1e-3:
                sleep(sleep_period)
```

and check if it gets better? You have a very high message rate, i assume the `sleep()` call slows you down too much.

zariiii9003, 2022-08-14:
try this branch please: https://github.com/zariiii9003/python-can/tree/message_sync

zixuan-su, 2022-08-15:
> Could you replace this
> 
> https://github.com/hardbyte/python-can/blob/3a7c80eb24d253b092b038f45be9d36942c2ab2d/can/io/player.py#L154
> 
> with this
> 
> ```python
>             if sleep_period > 1e-3:
>                 sleep(sleep_period)
> ```
> 
> and check if it gets better? You have a very high message rate, i assume the `sleep()` call slows you down too much.

I replace 
"with _create_bus(results, **additional_config) as bus:
        with LogReader(results.infile, **additional_config) as reader:

            in_sync = MessageSync(
                cast(Iterable[Message], reader),
                timestamps=results.timestamps,
                gap=results.gap,
                skip=results.skip,
            )
"
with 
"Msg_List = list(MessageSync(results.infile))
       for message in Msg_List:
            ..........
"
It only takes 44 s to play a 120 s BLF file  
[image]

Then I will try your suggestion

zixuan-su, 2022-08-15:
> try this branch please: https://github.com/zariiii9003/python-can/tree/message_sync

I follow your suggestion,that`s cool  !!!   
[image]

zariiii9003, 2022-08-15:
Thank you for your feedback, then i'll merge it into develop.

PR #1374 (2022-08-15): Improve MessageSync timings
Close #1372
closed 2022-08-15 by zariiii9003