hardbyte/python-can #1378: Fix the compressed BLFWriter and compressed SizedRotatingLogger rollover

j-c-cook, 2022-08-22:
### Describe the bug

The compressed BLFWriter (`file.blf.gz`) is not currently functional. I have made two commits ahead of https://github.com/hardbyte/python-can/tree/develop in https://github.com/j-c-cook/python-can/tree/Fixblfgz that address the problem. They are described below:
- The default setting for the `compress`ed logger is to append, and the `self.file.seek(0)` function call in `BLFWriter`s method, `stop(self)` throws an OSError. The commit, https://github.com/j-c-cook/python-can/commit/9259ec09c4b3ae78c05521551e6a2f1cf6aba0b3, now fixed the `file.blf.gz` logging by defaulting to overwrite, rather than append (Why is overwrite `w` default everywhere except for in the case of `compress`?).
- The rollover name creation did not adequately create files. A rolled over file would look something like: `file.blf.time#0000.gz`. The rollover filename creation was fixed in https://github.com/j-c-cook/python-can/commit/5b9b49a0a24c140fee8e30831edbfebcb0eec2d6. 

### To Reproduce

Enable virtual can: `sudo ip link add dev vcan0 type vcan && sudo ip link set up vcan0`
Spam the virtual bus with random can messages 1 ms apart: `cangen vcan0 -e -I 18EAFF26 -L i -v -v -v -g 1`
Log the virtual bus: `python -m can.logger -c vcan0 -i socketcan -b 250000 -f file.blf.gz -s 150000`

### Expected behavior

If the append option, `-a` is not described, then the `compress` function should not be trying to append to an existing file. The rolled over log files should be `Logger` compatible. Still outstanding, is how to make the append option work with `compress`. 

### Additional context

OS and version: Ubuntu 22.04.4 LTS
Python version: Python 3.7.17
python-can version: https://github.com/hardbyte/python-can
python-can interface/s (if applicable):

christiansandberg, 2022-08-30:
Should be fixed somehow but it is also quite unnecessary to gzip a blf file since it already uses compression internally.

j-c-cook, 2022-08-30:
> ... it is also quite unnecessary to gzip a blf file since it already uses compression internally.

This is something that would be useful to quantify. I am using secure copy protocol, `scp`, to transfer CAN logs from a remote device to a server. I have ensured that the files being transferred are zipped due to some previous experiences (empiricism). 
I have found in the past when transferring many files or even just one large file with `scp` that a zipped file is transported much faster. I am curious though now that you say this, in the case of `blf` versus `blf.gz`, what the percent difference in transfer time is.

PR #1429 (2022-11-13): Add gzip check to compress method
Seeks to address @zariiii9003's request from https://github.com/hardbyte/python-can/pull/1385#issuecomment-1297372655.

closes #1378
closed 2022-11-13 by zariiii9003