hardbyte/python-can #1380: Question

Sharmaranjan, 2022-08-24:
Currently python-can extracts can data only from any blf, what it does to ethernet data?
And on what parameters it is skipping?

If possible can you suggest any way or library which can extract ethernet data as well

zariiii9003, 2022-08-26:
This library probably supports Ethernet data: https://bitbucket.org/tobylorenz/vector_blf/src/master/
But it's C++.

Sharmaranjan, 2022-08-26:
I knew about this c++ Library, but anyway thanks.

I am looking for any python Library

felixdivo, 2022-11-12:
The supported interfaces of this library are listed [here](https://python-can.readthedocs.io/en/master/interfaces.html). You can also interface with C++ code from python.

felixdivo, 2022-11-12:
I think there is no clear task to complete in this issue.

closed 2022-11-12 by felixdivo