hardbyte/python-can #1381: Suggest engage the Home Assistant community to encourage an integration for python-can and other CAN Bus OBD-II/OBD2 adapters

Hedda, 2022-08-25:
### Is your feature request related to a problem? Please describe.

This is a feature request of sort as a request to enable an integration for Home Assistant that would depend on python-can.

### Describe the solution you'd like

I would like to suggest and recommend that a python-can developer(s) try to engage with the Home Assistant community to encourage a Home Automation "integration" (also referred to as "integration component" and "custom component") for python-can protocol/API specifically and/or a generic integration for CAN Bus OBD-II/OBD2 (OBD/EOBD+CANbus) adapters which supports the same APIs and protocols for third-party integrations.

Check out feature request discussion -> https://community.home-assistant.io/t/support-for-can-bus-peak-ixxat-canable-etc/249639/

Home Assistant is an open-source home automation software application and is currently the world's most popular platform for home automation enthusiasts, primarily because it makes it relatively easy for developers to make third-party integrations for it so it, therefore, has more third-party integrations than other home automation framework, and it is also one of the most active Python-based projects on GitHub.

https://en.wikipedia.org/wiki/Home_Assistant

https://github.com/home-assistant/

https://www.home-assistant.io/

Home Assistant community members have requested support for different OBD-II/OBD2 adapters for a long time in order to both use it as a device tracker and/or presence detection of their cars and trucks (when it connects you to your Wi-Fi at home) as well as get data information about things like battery and fuel level status which is useful in home automation use cases like sending status notifications when the car/truck arrive at home or when battery or fuel is low.

See example:

https://developers.home-assistant.io/docs/development_index/

https://developers.home-assistant.io/docs/dev_101_services/

https://developers.home-assistant.io/docs/creating_component_index/

https://www.youtube.com/watch?v=e3VwPb72Bbg

https://www.youtube.com/watch?v=6LHeoUS1R40

https://www.youtube.com/watch?v=_yVzgrGt1ok

Here are by the way some somewhat similar sensor integrations for HA, most of these only offer "sensors" and "binary sensors" but some of these have offered some type of device tracker and/or presence detection as well:

https://www.home-assistant.io/integrations/#car

https://www.home-assistant.io/integrations/torque/

https://www.home-assistant.io/integrations/bmw_connected_drive/

https://www.home-assistant.io/integrations/dacia/

https://www.home-assistant.io/integrations/mazda/

https://www.home-assistant.io/integrations/nissan_leaf/

https://www.home-assistant.io/integrations/subaru/

https://www.home-assistant.io/integrations/volvooncall/

https://brands.home-assistant.io/_/nsw_fuel_station/logo.png

https://www.home-assistant.io/integrations/openevse/

https://www.home-assistant.io/integrations/juicenet/

https://www.home-assistant.io/integrations/traccar/

https://www.home-assistant.io/integrations/tractive/

https://www.home-assistant.io/integrations/owntracks/

https://www.home-assistant.io/integrations/fleetgo/

https://www.home-assistant.io/integrations/mqtt_room/

https://www.home-assistant.io/integrations/device_sun_light_trigger/

https://www.home-assistant.io/integrations/asuswrt/

There are also more integrations for other non-car related bus protocols which present sensors and binary sensor, etc. entities to Home Assistant, ex:

https://www.home-assistant.io/integrations/velbus/

https://www.home-assistant.io/integrations/ubus/

https://www.home-assistant.io/integrations/modbus/

Personally, I would love to have a new integration which combines sensors, binary sensors, and device tracker and/or presence detection (person tracking) for knowing vehicles location prioritized if there is  developer interest in that:

https://www.home-assistant.io/integrations/device_tracker/

https://www.home-assistant.io/integrations/#presence-detection

https://www.home-assistant.io/integrations/person/

https://www.home-assistant.io/integrations/proximity/

https://www.home-assistant.io/integrations/ping/

https://www.home-assistant.io/integrations/zone/

### Describe alternatives you've considered

Engage the Home Assistant community to encourage an integration for python-can and other CAN Bus OBD-II/OBD2 adapters

### Additional context

zariiii9003, 2022-09-11:
The home-assistant users should encourage the home-assistant maintainers.
Unless there is some problem with the current python-can implementation, i don't know what you expect us to do. Nothing is stopping them from using python-can.

Hedda, 2022-09-12:
@zariiii9003 My point of this suggestion was only to recommend that some existing python-can developers might potentially be interested in independently developing an independent Can Bus integration (based on python-can) for Home Assistant.

I believe the end-user target audience is there and having such an integration would probably benefit both communities, as if there was a such integration then it could help increase the userbase of both projects, which in turn could maybe help you gain more developers as well in the long run.

However, with Home Assistant today being the world's popular home automation application and the largest Python-based project on GitHub,  with 2000+ integrations listed as officially supported + countless unofficial integrations not listed, I do not expect the Home Assistant founders/maintainers (i.e. [Nabu Casa employed developers](https://www.nabucasa.com/about/)) to step-up write and maintain their own officially supported first-part integrations for this as I think they core more on their place in maintaining Home Assistant core and the largest officially supported integrations, e.g. MQTT, ESPHome, HomeKit, Z-Wave, Zigbee, etc. -> https://analytics.home-assistant.io/#integrations

Anyway, I know that Home Assistant founders/maintainers generally try to encourage independent developers and third parties to write custom integrations for Home Assistant. Such third-party integrations are usually posted by independent developers via "HACS" (Home Assistant Community Store).

https://hacs.xyz

But, if an independently developed integration covers a common use case scenario in a standard matter and grows in popularity then it can happen that Home Assistant founders/maintainers contact the developer and ask if they like to publish it to the officially listed integrations -> https://www.home-assistant.io/integrations/

Home Assistant founders/maintainers do however maintainers the core APIs and helper + services integrations with the aim to every year make it easier for independent developers to write integrations for third-party use cases.

https://developers.home-assistant.io

Just my two cents.