hardbyte/python-can #1386: Getting started

HussainMughal1, 2022-09-02:
### Describe the bug
Hello. What is the process for getting started with this library. I've been struggling to import this library in my python project. 

### To Reproduce
Downloading this library and attempting to import into an editor.

### Expected behavior
Import python-can

### Additional context

OS and version: Windows 10
Python version: 3.7.3
python-can version: 4.0.0
python-can interface/s (if applicable): PCAN-USB Dongle is the intended use.

I've installed python and python-can. Do I need to copy the python-can repo into Program Files or anywhere else? Have I missed a step in the installation process?

<details><summary>Traceback and logs</summary>

</details>

j-c-cook, 2022-09-02:
See the [example](https://github.com/hardbyte/python-can#example-usage) in the README. While the name of the project is `python-can`, the name of the package is `can`. 

Install to pip: `pip install python-can`
Import in module: `import can`

This appears to be a `question`, not a bug report.

HussainMughal1, 2022-09-06:
Hello, 

The installation process was fine. I figured that I wasn't able to import the package into an existing project, but I could use it with any newly created project post-installation. 

Thank you

closed 2022-09-06 by HussainMughal1