hardbyte/python-can #1391: Vector Hardware Question

dhje0ng, 2022-09-13:
Hello sir. I'm trying to inject data into a CAN bus via python-can.
Since I am connecting two VN5620 physical hardware to the PC, I think the bus connection method I think is channel 0, 1, 2, 3.

but. Only channels 0 and 1 are available, and an error occurs for the remaining channels.
`can.interfaces.vector.exceptions.VectorInitializationError: xlGetApplConfig failed (XL_ERROR) [Error Code 255]`

Can you please tell me why I am getting this error?

i tried this code.

```py
bus_channel = 0
#bus = VectorBus(bus_channel, bitrate=500000, data_bitrate=2000000, fd=True, receive_own_messages=True)
or
bus = can.interface.Bus(bustype='vector', channel=bus_channel, bitrate=500000, data_bitrate=2000000, fd=True, receive_own_messages=True)
```
[image]

zariiii9003, 2022-09-13:
The default parameter for `app_name` is `"CANalyzer"`. Either you assign the CANalyzer app to your target channel or you can use the `serial` argument instead.

dhje0ng, 2022-09-13:
thank you sir. this problem is solved.

closed 2022-09-13 by dhje0ng