hardbyte/python-can #1393: different output between msg and msg.data

muggel042, 2022-09-13:
### Describe the bug
Hello

Im trying to communicate with a can device via can0. Currently I am not understanding, how there can be a difference between output of the full message, after receiving data via recv() and only outputting msg.data

### To Reproduce
```python
bus = can.interface.Bus(channel = 'can0', interface='socketcan')
command = "C1 00 0F 00 00 00 00"
msg_data = [int(r, 16) for r in command.split(" ")]
receiver_id = int("19", 16)
msg = can.Message(arbitration_id=receiver_id, data=msg_data, is_extended_id=False, dlc=7)
bus.send(msg)
rc = bus.recv(0.05)

print(rc)
print(rc.data)
```
### Expected behavior
Output:
Timestamp: 1663091801.003547        ID: 0600    S Rx                DL:  7    02 19 0f **01 48** 00 00        Channel: can0
bytearray(b'\x02\x19\x0f **\x01H** \x00\x00')

I'd expect the output to be identical for rc and rc.data

### Additional context

OS and version: Raspbian GNU/Linux 10 (buster)
Python version: 3.7
python-can version: 4.0.0
python-can interface/s (if applicable):

Best regards

zariiii9003, 2022-09-13:
It is just the way python prints `bytes` objects. Some bytes are represented by the corresponding ASCII character: 0x48 => H

muggel042, 2022-09-13:
Thanks. I got a lot to learn then :-)

closed 2022-09-13 by muggel042