hardbyte/python-can #1398: How connect USB-CAN Adapter

sasha1396, 2022-09-19:
Hello there! Very nice code, but I still haven't been able to try it. Can you please tell me how can I use canopen with USBCAN adapter? My adapter is defined as a single USB device:

[image]

It looks like this:
[image]
[image]
[image]

And I still have not figured out which interface to use for it and how. I would appreciate any help and directions to the right place.

j-c-cook, 2022-09-25:
What is `canopen`? Have you read the `README` and docs? Have you installed a driver for your device? I have seen issues in recent discussing a function that can `find_all_interfaces`. You may want to see if that issue is completed and see if `python-can` autodetects your bus (device and interface). 

Note: This issue is improperly marked as an enhancement.

sasha1396, 2022-09-26:
I read all the information I could find. As I understand it, none of the interfaces is suitable. The only thing I decided to try is Cananalyst-II, since it is most suitable for my device, but I could not go further than connecting to the device, I get an error: canopen.nmt.NmtError: Timeout waiting for boot-up message.
I did not see the find_all_interfaces function in any of the repositories. So I don't know what to do next.

zariiii9003, 2022-09-26:
`can.detect_available_configs()` might help.

sasha1396, 2022-09-26:
> `can.detect_available_configs()` might help.

I used this: **can.detect_available_configs()**  What should I do about it?
Kvaser canlib is unavailable.
fcntl not available on this platform
socket.CMSG_SPACE not available on this platform
failed to fetch opened can devices: [WinError 2] Cannot find the specified file
You won't be able to use the serial can backend without the serial module installed!
Cannot load IXXAT vcinpl library: Could not find module 'vcinpl.dll' (or one of its dependencies). Try using the full path with constructor syntax.
Cannot load IXXAT vcinpl library: Could not find module 'vcinpl2.dll' (or one of its dependencies). Try using the full path with constructor syntax.
Failed to load NI-CAN driver: Could not find module 'nican' (or one of its dependencies). Try using the full path with constructor syntax.
Failed to load IS-CAN driver: Could not find module 'iscandrv' (or one of its dependencies). Try using the full path with constructor syntax.
You won't be able to use the ICS neoVI can backend without the python-ics module installed!: No module named 'ics'
Using ICS neoVI can backend without the filelock module installed may cause some issues!: No module named 'filelock'
Could not import vxlapi: Could not find module 'vxlapi64' (or one of its dependencies). Try using the full path with constructor syntax.
You won't be able to use the slcan can backend without the serial module installed!
You won't be able to use the Robotell can backend without the serial module installed!
Cannot load SYSTEC ucan library: Could not find module 'usbcan64.dll' (or one of its dependencies). Try using the full path with constructor syntax..
The SYSTEC ucan library has not been initialized.
You won't be able to use the serial can backend without the serial module installed!
The CANtact module is not installed. Install it using `python -m pip install cantact`

zariiii9003, 2022-09-28:
Seems like your interface is not supported.

sasha1396, 2022-09-28:
> Seems like your interface is not supported.

What else can I try?

j-c-cook, 2022-09-29:
A  "low cost open-source USB to CAN adapter" is [CANable 2.0](https://openlightlabs.com/collections/frontpage/products/canable-2-0). However, they are out of stock.

You might want to look into [can/interfaces](https://github.com/hardbyte/python-can/tree/develop/can/interfaces). I have never written an interface, so that's all I know. 

Please note that this package is called `python-can`. It is not clear what you mean by "CanOpen". 

If you are going to write an interface, this issue should either be renamed to "Create new interface for xxx device" or should be closed and a new issue described.