hardbyte/python-can #1400: can.viewer hangs when encountering CANFD data frames

o92, 2022-09-28:
### Description
can.viewer stops displaying incoming messages and the screen hangs upon encountering CAN messages that are sent using flexible data rate mode. 

### Expected behavior
The expected behavior should be that the can viewer must not crash or hang. It should display the CAN message even if DLC > 8

### Additional context

OS and version: Windows 10 Enterprise
Python version: 3.8.10
python-can version: 4.0.0
python-can interface/s (if applicable): Vector VN1630A

While running can.viewer on one instance by executing as follows:
> python -m can.viewer -c 1 -i vector -b 500000 --fd

I try to pump in some arbitrary messages using parameters for flexible data rate. The following is my simple script for replication. 

```
import can
import time

print('\n\rCAN Tx test')

try:
	bus = can.interface.Bus(channel='1', bustype='vector', fd = True, bitrate=500000)
except Exception as e:
	print(e)
	exit()
	

# Main loop
while 1:
     msg = can.Message(arbitration_id=0x012,is_fd = True, bitrate_switch = True, data=[0,0,0,0,0,0x1e,0x21,0xfe, 0x80, 0, 0,0, 0],is_extended_id=True)
     bus.send(msg)
     time.sleep(0.2)

```

The can.viewer hangs upon receiving a frame whose DLC > 8.

zariiii9003, 2022-09-28:
Have you tried the develop branch? The can.viewer works fine with CAN FD and long messages, i just tested it.

o92, 2022-09-29:
@j-c-cook @zariiii9003 First of all, thank you very much for your response. I got myself the dev branch installed and tried out. 

However, when I execute the script with > python -m can.viewer -c 1 -i vector, I can see the messages but they are truncated. The DLC indication is correct. Note that, I did have to supply --fd flag. If I did that, I couldn't see the messages that I was sending with DLC> 8.

[image]

zariiii9003, 2022-09-29:
Can you run this?

```python3
import curses
import random
import threading
import time

import can
from can.interfaces.vector import VectorBus
from can.viewer import CanViewer

def send_fd_messages(bus: VectorBus) -> None:
    while True:
        length = random.randint(1, 64)
        data = random.randbytes(length)
        msg = can.Message(
            arbitration_id=length,
            data=bytearray(data),
            is_fd=True,
            is_rx=False,
            bitrate_switch=True,
            dlc=length,
        )
        bus.send(msg)
        time.sleep(0.1)

if __name__ == '__main__':
    bus = VectorBus(
        serial=100,  # the serial number of your virtual bus. Can be 0 or 100
        channel=0,
        fd=True,
        bitrate=500_000,
        data_bitrate=1_000_000,
        receive_own_messages=True,
    )
    send_thread = threading.Thread(target=send_fd_messages, args=(bus,), daemon=True)
    send_thread.start()

    curses.wrapper(CanViewer, bus, None)

```

o92, 2022-09-30:
@zariiii9003 I can see the CAN FD frames when I run your sample script. However, it is not what I want because the serial=100 specifies to use the virtual bus instead. I am not testing on virtual bus. I am doing it on the actual bus with a real ECU connected. Therefore, I still cannot see the CAN FD messages on the physical bus.

o92, 2022-09-30:
The following is your modified code which I used to test on my setup. The python version that I use is 3.8, therefore, random.randbytes is not supported. os.urandom replaces that line. 

```
import curses
import random
import threading
import time
import os 

import can
from can.interfaces.vector import VectorBus
from can.viewer import CanViewer

def send_fd_messages(bus: VectorBus) -> None:
    while True:
        length = random.randint(1, 10)
        data = os.urandom(length)
        msg = can.Message(
            arbitration_id=length,
            data=bytearray(data),
            is_fd=True,
            is_rx=False,
            bitrate_switch=True,
            dlc=length,
        )
        bus.send(msg)
        time.sleep(0.2)

if __name__ == '__main__':
    bus = VectorBus(
        #serial=100,  # the serial number of your virtual bus. Can be 0 or 100, 
        channel=1,
        fd=True,
        bitrate=500_000,
        data_bitrate=2_000_000,
        receive_own_messages=True,
    )
    send_thread = threading.Thread(target=send_fd_messages, args=(bus,), daemon=True)
    send_thread.start()

    curses.wrapper(CanViewer, bus, None)

```

o92, 2022-09-30:
Just an update:

I can now see the CAN FD messages from the ECU, but I cannot see the messages that I sent out. The script is as follows:

```
import random
import time
import os 
import can
from can.interfaces.vector import VectorBus

def send_fd_messages(bus: VectorBus) -> None:
    while True:
        length = random.randint(1, 12)
        data = os.urandom(length)
        msg = can.Message(
            arbitration_id=0,
            data=bytearray(data),
            is_fd=True,
            is_rx=False,
            bitrate_switch=True,
            dlc=length,
        )
        bus.send(msg)
        time.sleep(0.1)

if __name__ == '__main__':
    bus = VectorBus(
        app_name='python-can',  # the serial number of your virtual bus. Can be 0 or 100
        channel=1,
        fd=True,
        bitrate=500_000,
        data_bitrate=2_000_000,
        receive_own_messages=True,
    )
    send_fd_messages(bus)
    
```

o92, 2022-09-30:
Also, the moment I initialize Kvaser device in code, it throws the following error. 

```
import random
import time
import os 
import can
#from can.interfaces.vector import VectorBus
from can.interfaces.kvaser import KvaserBus

def send_fd_messages(bus: KvaserBus) -> None:
	while True:
            print(1)
		# length = random.randint(1,8)
		# data = os.urandom(length)
		# msg = can.Message(arbitration_id=0x18DBFFF9,data=bytearray(data),is_fd=False,is_rx=False,bitrate_switch=False,dlc=length)
		# print(msg)
		# bus.send(msg)
		# time.sleep(0.5)

if __name__ == '__main__':
    bus = KvaserBus(
        #app_name='python-can',  # the serial number of your virtual bus. Can be 0 or 100
        channel=1,
        fd=True,
        bitrate=500_000,
        data_bitrate=2_000_000,
        receive_own_messages=False,
        
    )
    send_fd_messages(bus)

```

## Traceback 
Traceback (most recent call last):
  File ".\fd_tx_test.py", line 21, in <module>
    bus = KvaserBus(
  File "C:\N92\python-can dev\lib\site-packages\can\interfaces\kvaser\canlib.py", line 453, in __init__
    self._read_handle = canOpenChannel(channel, flags)
  File "C:\N92\python-can dev\lib\site-packages\can\interfaces\kvaser\canlib.py", line 131, in __check_bus_handle_validity
    raise CANLIBInitializationError(function, result, arguments)
can.interfaces.kvaser.canlib.CANLIBInitializationError: Function canOpenChannel failed - Specified device not found [Error Code -3]

## Kvaser Device guide 

[image]

o92, 2022-10-03:
Hello, 

Thank you both @j-c-cook @zariiii9003 for your help. I have managed to found out that the Kvaser VCI device that I have, only supports CAN and not CAN FD. As for the Vector device, our ECU has some special set up that only can communicate with specific peripheral devices. When I connected such a peripheral unit, I could read and send CAN FD messages directly on the CAN bus. And it works with both the older version of can.viewer as well as the dev branch. 

I will close this case as it has been resolved for me.

j-c-cook, 2022-10-03:
Thank you for the update and closing your issue now that it's resolved. Glad to know the `develop` branch is functional for your application. 🚀

zariiii9003, 2022-10-04:
> Hello,
> 
> As for the Vector device, our ECU has some special set up that only can communicate with specific peripheral devices. When I connected such a peripheral unit, I could read and send CAN FD messages directly on the CAN bus. 

That doesn't sound plausible to me. I assume that you set your bit timings correctly.
Could it be that you are using the non ISO CANFD version? You could add a new line here to try that:
https://github.com/hardbyte/python-can/blob/89c395fd315179b1eb2ca8e21a852c825bb385b8/can/interfaces/vector/canlib.py#L399

```python3
        canfd_conf = xlclass.XLcanFdConf()
        canfd_conf.options = xldefine.XL_CANFD_ConfigOptions.CANFD_CONFOPT_NO_ISO.value
```