hardbyte/python-can #1402: SizedRotatingLogger fails with .BLF not with .log

skipper85, 2022-09-30:
### Describe the bug
Exception in Notifier when BLF file is closed for rotation, but new CAN messages are written to the closed BLF file.

### To Reproduce
Happens at each rotation with BLF file, does not happen with the log extension.

### Expected behavior
No Exception and new BLF logger file

### Additional context

OS and version: Windows 10
Python version: 3.10.7
python-can version: 4.0.0
python-can interface/s (if applicable):

<details><summary>Traceback and logs</summary>

```python
Exception in thread can.notifier for bus "Application CANalyzer: CAN 1":
Traceback (most recent call last):
  File "C:\Users\Public\AppData\Local\Programs\Python\Python310\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\Users\Public\AppData\Local\Programs\Python\Python310\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Public\Documents\env\lib\site-packages\can\notifier.py", line 121, in _rx_thread
    self._on_message_received(msg)
  File "C:\Users\Public\Documents\env\lib\site-packages\can\notifier.py", line 143, in 
_on_message_received
    res = cast(Union[None, Optional[Awaitable[Any]]], callback(msg))
  File "C:\Users\Public\Documents\env\lib\site-packages\can\listener.py", line 43, in __call__
    self.on_message_received(msg)
  File "C:\Users\Public\Documents\env\lib\site-packages\can\io\logger.py", line 205, in 
on_message_received
    self.do_rollover()
  File "C:\Users\Public\Documents\env\lib\site-packages\can\io\logger.py", line 341, in do_rollover
    self._writer = self._get_new_writer(self.base_filename)
  File "C:\Users\Public\Documents\env\lib\site-packages\can\io\logger.py", line 221, in _get_new_writer
    self._writer.stop()
  File "C:\Users\Public\Documents\env\lib\site-packages\can\io\blf.py", line 565, in stop
    if self.file.seekable():
ValueError: I/O operation on closed file
```

```python
import logging
from model.can_bus import CanBus
from model.can_device import CanDevice
from model.settings import Settings
from model.can_device import CanDevice
from model.can_rx_worker import CanRxWorker
from model.can_tx_worker import CanTxWorker
from PyQt6.QtCore import QSize, Qt, QThreadPool

root_logger = logging.getLogger(name='main')
root_logger.setLevel(logging.DEBUG)

ch = logging.StreamHandler()   
fh = logging.FileHandler(f"C:/logging/log/logging_{Settings.get_datetime_str()}.log")

formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')

ch.setFormatter(formatter)
fh.setFormatter(formatter)

root_logger.addHandler(ch)
root_logger.addHandler(fh)

try:
    threadpool = QThreadPool()
    bus = CanBus(device_type="VECTOR", logger=root_logger)
    canbus = bus.initialize()

    can_devices = [None] * 6
    can_devices[0] = CanDevice(
                    canbus=canbus,
                    autosar_db_file="autosar.arxml",
                    xcp_db_file="xcp.a2l",
                    logger=root_logger)
    
    can_devices[0].initialize_xcp()
    can_devices[0].initialize_xcp_measurement()

    can_rx_worker = CanRxWorker(
                canbus=canbus,
                can_devices=can_devices,
                logger=root_logger,
                blf_filename=f"C:/data/logging_{Settings.get_datetime_str()}.blf"
            )
    
    threadpool.start(can_rx_worker)
    
    can_tx_worker = CanTxWorker(
                canbus=canbus,
                can_devices=can_devices,
                logger=root_logger
            )
    
    threadpool.start(can_tx_worker)
except Exception as ex:
    root_logger.exception(ex.args[0])
```

Partial code of the CanRxWorker(QRunnable) with the Notifier and Listener
```python
self.can_rx_logger = SizedRotatingLogger(
    base_filename=self.blf_filename,
    max_bytes=500 * 1024 ** 2,  # =500MB
)

self.can_rx_buffer = CanRxBuffer(
    canbus=self.canbus,
    can_devices=self.can_devices,
    logger=self.logger,
    signals=self.signals,
    can_data=self.can_data
)

# add listeners to the notifier
self.notifier = Notifier(
    self.canbus,
    [
        self.can_rx_logger,
        self.can_rx_buffer
    ]
)
```
</details>

zariiii9003, 2022-10-04:
That's fixed in #1317.

closed 2022-10-04 by zariiii9003

skipper85, 2022-10-05:
Ok, thank you for the responses, I will try out the develop branch.