hardbyte/python-can #1404: Problem using "import can" in PyCharm

Darkrem1, 2022-10-05:
Hi guys,

I'm a PLC programmer with just basic knowledge in Python so it is probably just something basic missing.

I've installed python-can in PyCharm. Usng the example "send_one", it acts like it is not installed. I've worked with classes and module but when I type "can. ..." nothing shows for auto-complete like it is not imported correctly

Gettings errors like:

Cannot find reference 'CanError' in 'imported module can' 

  File "C:\Users\rbeland\PycharmProjects\pythonProject\main.py", line 12, in send_one
    with can.interface.Bus() as bus:
AttributeError: module 'can' has no attribute 'interface'

Thanks

j-c-cook, 2022-10-05:
This problem could lie in your "Python Interpreter" selection in PyCharm, which could be verified by running your script from the command line. 

You can run the script from the command prompt by typing `python path/to/file.py` and pressing enter. If you did not install Python to your path, then you will need to provide the path to your Python executable to do so. Install the `python-can` package to pip from the command line first. 
`
python -m pip install python-can
`

zariiii9003, 2022-10-07:
Maybe you have another file named `can.py` or a package named `can`?

felixdivo, 2022-11-12:
> Maybe you have another file named `can.py` or a package named `can`?

I already had this happen several times on my machines ...

felixdivo, 2022-11-12:
I think we can assume it is solved by now.

closed 2022-11-12 by felixdivo