hardbyte/python-can #1414: SLCAN with TCP IP not supported

ElieSKickMaker, 2022-10-19:
### Describe the bug
The doc of slcan said "Remote serial ports will be specified via special URL. Both raw TCP sockets as also RFC2217 ports are supported." 
But it don't seems to be supported I've got the error : 
`can\interfaces\slcan.py", line 91, in __init__
    raise CanInterfaceNotImplementedError("The serial module is not installed")
can.exceptions.CanInterfaceNotImplementedError: The serial module is not installed`

My code is working with a real pcan usb. 

### To Reproduce

Init code of the bus :
self.bus = can.interfaces.slcan.slcanBus(
            channel='socket://127.0.0.1:4545'
            ) 

OS and version: Windows
Python version: 3.9.15
python-can version:4.0.0
python-can interface/s (if applicable):
slcan

zariiii9003, 2022-10-20:
[Install pyserial](https://pyserial.readthedocs.io/en/latest/pyserial.html#from-pypi)

ElieSKickMaker, 2022-10-24:
thanks for your help @zariiii9003

closed 2022-10-24 by ElieSKickMaker