hardbyte/python-can #1416: Broadcast Manager fails for large sets of messages

sd8f907889, 2022-10-24:
### Describe the bug

I am attempting to use Broadcast Manager with a PEAK USB CAN adapter on a Raspberry Pi (running the standard Pi OS distro). I am creating a fairly large List containing over 10000 can.Message instances, and would like to use the Broadcast Manager to send this list of messages (same arbitration ID, same DLC on all messages, just different data) in a one-shot manner at a specific time interval.

When I attempt to run my program, I get the following exception:

```bash
can.exceptions.CanOperationError: Couldn't send CAN BCM frame due to OS Error: Invalid argument You are probably referring to a non-existing frame. [Error Code 22]
```

However, if I reduce the size of my message List to something in the 100-200 CAN Frames range, the exact same code works just fine and I see the expected behavior from the device I am interacting with over CAN.

### To Reproduce

- Construct an empty list, and append more than 256 CAN frames
- Attempt to send this list of messages using Broadcast Manager send_periodic() task (my interval is in the range of a few milliseconds)
- To reliably reproduce this issue, make the list above 257 frames in length. It appears this is the boundary condition for this issue (at least in my experience... i.e. 256 frames can be handled by the BCM just fine but 257 frames cause the error mentioned above.)

Due to IP issues I would prefer not to share source. However, if the above is insufficient to reproduce the issue I am happy to write a self-contained code listing.

### Expected behavior

Transmission of all CAN frames placed in list at the specified interval

### Additional context

OS and version: Raspberry Pi OS (Kernel 5.15.61-v7 armv7l)
Python version: 3.9.2
python-can version: 4.0.0 (installed via Pip)
python-can interface/s (if applicable): SocketCAN (via PEAK PCAN-USB)

<details><summary>Traceback and logs</summary>

```bash
  File "/usr/local/lib/python3.9/dist-packages/can/bus.py", line 235, in send_periodic
    self._send_periodic_internal(msgs, period, duration),
  File "/usr/local/lib/python3.9/dist-packages/can/interfaces/socketcan/socketcan.py", line 827, in _send_periodic_internal
    task = CyclicSendTask(bcm_socket, task_id, msgs, period, duration)
  File "/usr/local/lib/python3.9/dist-packages/can/interfaces/socketcan/socketcan.py", line 350, in __init__
    self._tx_setup(self.messages)
  File "/usr/local/lib/python3.9/dist-packages/can/interfaces/socketcan/socketcan.py", line 374, in _tx_setup
    send_bcm(self.bcm_socket, header + body)
  File "/usr/local/lib/python3.9/dist-packages/can/interfaces/socketcan/socketcan.py", line 293, in send_bcm
    raise can.CanOperationError(base + specific_message, error.errno) from error
can.exceptions.CanOperationError: Couldn't send CAN BCM frame due to OS Error: Invalid argument You are probably referring to a non-existing frame. [Error Code 2```
</details>

hardbyte, 2022-10-25:
The linux kernel imposes a limit on the number of CAN frames that can be sent in one transmission task see - https://www.kernel.org/doc/html/latest/networking/can.html#broadcast-manager-message-sequence-transmission

However you should be able to create many separate transmission tasks.

sd8f907889, 2022-10-25:
Thanks a lot for the reply - Just looked at the link and that would certainly explain my indications :)

Any ideas on ways to accurately swap data buffers in with some kind of callback architecture? I briefly looked around the docs and can't seem to find anything brought out by the Python interface to make this possible in a deterministic manner. My current workaround would be to add a max time to the send_periodic call, set to 256*T_sample, then a sleep() of that amount before calling a modify_data method to put fresh data in the buffer, but I'd highly prefer a callback.

sd8f907889, 2022-10-25:
Also - While this limitation can be found in the Kernel docs, I would recommend updating the Broadcast Manager documentation to reflect this.

hardbyte, 2022-10-25:
I wouldn't be trying to swap the data between each transmission. Instead how about trying to create a separate transmission task for each message you want to send?

sd8f907889, 2022-10-25:
I concur with that sentiment but in that case, would I just have to manually kick off the individual transmit cyclic tasks and time things myself? Or is there some "end of cycle" callback I can hook into to dispatch my one-shot transmissions of 256 elements each?

hardbyte, 2022-10-26:
Oh I missed that you said _same arbitration ID, same DLC on all messages_, the transmission tasks are indexed by can id so separate transmission tasks wouldn't work at all.

You might be able to configure the send task with `TX_COUNTEVT` and then block on the BCM socket until `TX_EXPIRED` is sent (after sending 255 messages), then you could change the data out. That will require digging under the covers a bit.

sd8f907889, 2022-10-26:
Yeah I was fearing I'd probably need to use some lower-level raw sockets to deal with this. I'm assuming python-can doesn't expose enough of the under-the-hood stuff for me to use it for this task?

hardbyte, 2022-10-26:
Oh I'd say there is plenty of code that will help with talking BCM - https://github.com/hardbyte/python-can/blob/451ce48eacd529453c14ec59fe6bc5d142f2e65a/can/interfaces/socketcan/socketcan.py#L51-L481

hartkopp, 2022-10-26:
> Yeah I was fearing I'd probably need to use some lower-level raw sockets to deal with this. I'm assuming python-can doesn't expose enough of the under-the-hood stuff for me to use it for this task?

The BCM can send up to 256 frames in a sequence. And when you do not change the number of frames (`nframes`) you can always overwrite the content of those (256) frames at runtime. The next (frame) transmission would directly contain the updated data. When doing so, you can decide whether you want to maintain the current pointer into the array of frames or if you want to reset the array pointer to zero (flag `TX_RESET_MULTI_IDX`).

Can you elaborate on your use-case, why you would need this sending of _more_ than 256 frames in a row with a timer?

sd8f907889, 2022-10-28:
I just have some data I am sending that requires low timing jitter, so manually timing it in a loop in the userspace would be suboptimal. Can't really provide any further detail unfortunately.

hartkopp, 2022-10-28:
Yes, I got that. But is this a continuous stream of CAN frames or a fix sequence of (e.g. 450) CAN frames?
Or is it just some segmented content that could be transferred over a CAN transport protocol like ISO-TP (ISO15765-2)?
Maybe I have a pretty weird idea for your problem: You might use the Linux traffic control tool `tc` together with a token bucket filter (TBF) to limit the bandwidth/speed for a specific CAN ID

sd8f907889, 2022-10-31:
It's a large continuous stream of CAN messages where the DLC, Arb ID and other attributes do *not* change, only the data payload. And said stream of messages has to be sent with deterministic latency between messages, as it relates to control of some physical hardware. The issue isn't really one of traffic control (the bus barely has 5% load)... it's more of a hard message interval requirement for reasons I can't really elaborate on further.

hardbyte, 2022-11-15:
It would also help with sending messages where the data changes frequently e.g. https://stackoverflow.com/questions/73559087/python-can-implement-a-message-counter

hartkopp, 2022-11-15:
But this example from StackOverflow is exactly implemented with the Linux CAN_BCM.
The referenced 4 bit counter is (commonly) placed inside the CAN frame data (e.g in data[1]) and the entire 16 CAN frame block can be _updated at once_ with (another) TX_SETUP message. Depending on the given flags this has no impact on the ongoing cyclic transmission cycle of the 16 frames sequence. Whenever the content has to be changed, it is changed in all 16 frames in the same way (preserving the 4 bit value for the counter). The update with TX_SETUP is an atomar write to the array of 16 CAN frames. The next TX timer trigger will automatically pick the latest content for sending.