hardbyte/python-can #1417: CANalyst-II can't receive replay data

alinasr-esf, 2022-10-25:
when i send message with this library i need to receive reply but i couldn't find any function to do that.

zariiii9003, 2022-10-27:
Are you looking for the [recv](https://python-can.readthedocs.io/en/4.0.0/bus.html#can.BusABC.recv) method?

j-c-cook, 2022-11-03:
> when i send message with this library i need to receive reply but i couldn't find any function to do that.

I am having trouble understanding what you need. @alinasr-esf Could you please elaborate?

alinasr-esf, 2022-11-03:
i want to use CANalyst-II library and as you know it have receive function 
you can see it here  [https://pypi.org/project/canalystii/](url)
`dev = canalystii.CanalystDevice(bitrate=500000)

# Receive all pending messages on channel 0
for msg in dev.receive(0):
    print(msg)`  
but i got null answer

alinasr-esf, 2022-11-03:
let me write code here 
i use this code to send and receive data with canalystii usb can converter (i want to receive answer of sending data)
`dev = canalystii.CanalystDevice(bitrate=500000)`
`new_message = canalystii.Message(can_id=0x300, data_len=8, data=(0x01,0x02,0x03,0x04,0x05,0x06,0x07,0x08))`
`dev.send(0, new_message)`
```
for msg in dev.receive(0):
    print(msg)
```
`dev.stop(0)`

zariiii9003, 2022-11-04:
ping @projectgus

projectgus, 2022-11-04:
Hi @alinasr-esf,

Sorry that you're having trouble making this work. A few points that might be helpful:

1. This is not actually python-can, this is the [library that I made](https://github.com/projectgus/python-canalystii) that sits under python-can to support canalystii. If you want, you can install the actual [python-can library](https://python-can.readthedocs.io/en/master/installation.html) with canalystii as the backend. python-can has a lot more features and is easier to work with for many things.
2. When you call `dev.receive(0)` it returns all of the messages received until then, but it doesn't block to wait for any messages. So you might need to do something like:
  ```
  dev.send(0, ...)
  time.sleep(0.5)
  print(dev.receive(0))
  ```
  to give the other side time to respond to your message. If you use python-can instead, this part is much easier because [recv() has a timeout](https://python-can.readthedocs.io/en/master/bus.html#can.BusABC.recv).
3. Is it possible there is a problem with your CAN bus, i.e. baud rate mismatch or termination problem? If it is, no messages will be sent but canalystii hardware does not make it easy to detect errors.
4. It is possible that there is a bug in the python-canalystii library, or that the canalystii firmware has changed and is no longer compatible. But suggest trying out suggestions 1, 2, 3 first.

Good luck!

closed 2022-11-10 by alinasr-esf