hardbyte/python-can #1419: gs_usb interface incorrectly parses the is_extended_id flag

ryedwards, 2022-10-27:
The implementation of the gs_usb interface has a bug where the is_extended_id flag is not properly mapped from the gs_usb frame.

As-is:
```python
        frame = GsUsbFrame()

        # Do not set timeout as None or zero here to avoid blocking
        timeout_ms = round(timeout * 1000) if timeout else 1
        if not self.gs_usb.read(frame=frame, timeout_ms=timeout_ms):
            return None, False

        msg = can.Message(
            timestamp=frame.timestamp,
            arbitration_id=frame.arbitration_id,
            is_extended_id=frame.can_dlc,
            is_remote_frame=frame.is_remote_frame,
            is_error_frame=frame.is_error_frame,
            channel=self.channel_info,
            dlc=frame.can_dlc,
            data=bytearray(frame.data)[0 : frame.can_dlc],
            is_rx=True,
        )

        return msg, False
```
The line:
```python
            is_extended_id=frame.can_dlc,
```
Should be:
```python
            is_extended_id=frame.is_extended_id,
```

zariiii9003, 2022-10-27:
That was fixed in #1270. But it is only available in the develop branch for now.

ryedwards, 2022-10-27:
OK.  I reviewed the change there and it looks correct.  Closing this issue.

closed 2022-10-27 by ryedwards