hardbyte/python-can #1428: __init__.py empty

ygerlach, 2022-11-09:
### Describe the bug
The `__init__.py` file in `/usr/local/lib/python3.8/dist-packages/can/` seems to be empty, which results in interesting problems.

### To Reproduce
I installed this with `sudo pip3 install python-can==3.3.4` without errors.

### Expected behavior
the `__init__.py` should contain this: https://raw.githubusercontent.com/hardbyte/python-can/release-3.3.4/can/__init__.py

### Additional context

OS and version: Ubuntu 20.04
Python version: 3.8.10
python-can version: 3.3.4

felixdivo, 2022-11-12:
I tried to replicate it in the container `docker run -it python:3.8 /bin/bash` where I ran `pip3 install python-can==3.3.4`. Then it is in `site-packages`, i.e. in `/usr/local/lib/python3.8/site-packages/can/__init__.py` and has the correct content.

Can you replicate it in a container? If not, then it is definitely interesting, but most likely "just" a problem with your local setup. Still an issue, sure, but probably nothing where we can help much.

felixdivo, 2022-11-12:
Besides, there is also version `4.0.0` which you will most like want to switch over to anyways. 😄

closed 2022-11-14 by felixdivo