hardbyte/python-can #1432: bit timing is invalid in kvaser

zhikangfan01, 2022-11-15:
I used kvaser to customize the baud rate and found that the tseg values obtained in kwargs are all 0

python-can version:V4.0

This is not the case with v3.3.4

[image]

Could you please check the problem? Or could you please provide a reference code? Thank you

zariiii9003, 2022-11-16:
Could you print `kwargs`?

zariiii9003, 2022-11-16:
I guess the problem is here:
https://github.com/hardbyte/python-can/blob/2e2f157eb02c0505c4e38ca8cbbc54433c65351c/can/util.py#L250

Your timings are replaced with a `can.BitTiming` object, but most interfaces do not support that.

@christiansandberg Could you take a look here? Are there any downsides to keeping tseg1 etc. in the config?

zhikangfan01, 2022-11-19:
May I ask what content to modify to support the current use

zariiii9003, 2022-11-20:
You could instantiate KvaserBus directly 
```
from can.interfaces.kvaser import KvaserBus

bus = KvaserBus(channel=0, ...)
```

zhikangfan01, 2022-11-30:
Use the method you provided first to find or not to use@zariiii9003 
[image]

closed 2022-12-21 by zariiii9003