hardbyte/python-can #1448: Consider using __del__

felixdivo, 2022-11-24:
This was brought up in  #1435. We could implement `__del__` in the `BusABC` to do a best-effort cleanup and call `shutdown()` in case the user forgot to. We could also print a warning in such a case to notify the user of the improper use of the library. It is not guaranteed that the method is called, but if it is in some cases we might as well use it.

zariiii9003, 2022-11-24:
Or [weakref.finalize](https://docs.python.org/3/library/weakref.html#weakref.finalize). Can we assume that `shutdown` can be called multiple times without raised exceptions?

Tbruno25, 2023-01-04:
I'll go ahead and take this issue!

felixdivo, 2023-01-05:
> Can we assume that `shutdown` can be called multiple times without raised exceptions?

Nope, I don't think so. We might want to change that?

Tbruno25, 2023-01-16:
After reading through #1435 and looking at this closer, I'm not sure what value a `__del__` method will add.

It seems `BusABC` is intended be used solely as a context manager since it lacks the traditional `open` and `close` methods. Once the context is entered, I was unable to find a scenario through testing where `__exit__` and therefore `shutdown()` does not ultimately get called. 

I suppose if a user were to manually call `__enter__` then it could be applicable -- however I don't feel we should protect against the improper use of dunder methods.

Thoughts?

felixdivo, 2023-01-16:
The Bus class is more like [RAII](https://en.wikipedia.org/wiki/Resource_acquisition_is_initialization), if you will, in that creating the object opens the bus an prepares it for use. And they do have a `shutdown()`, which is just like a `close()`.

felixdivo, 2023-01-16:
So it can be used outside of a context manager, and that can be useful. For example, I have written a [ROS](https://docs.ros.org/en/rolling) node that acts as an interface to a CAN network. There, the "server" that handles the communication between the two network types holds a very long-lived bus instance and doing it with a context manager would at least make it more complicated and weird.

Although, yes: Many, many cases involve a context manager and if you can use it, you probably should. :)

Tbruno25, 2023-01-16:
Ahh got it. I should have read the docstrings closer :facepalm: 

Makes sense how `__del__ ` is beneficial in this case -- PR incoming shortly.

PR #1489 (2023-03-30): Add `__del__` method to `BusABC`
Initial PR to address #1448

Closes #1448
closed 2023-03-30 by zariiii9003