hardbyte/python-can #1453: CanInitializationError due to reported can_op_mode = 0

ntstimio, 2022-11-29:
### Describe the bug
The CAN bus (VectorBus) cannot be created because the can_op_mode value reported by the Vector CANcaseXL driver is set to 0, which is not an expected value.

### To Reproduce
Connect the CANcaseXL to a CAN bus.
Configure the CANcaseXL using Vector HardwareConfiguration tool.
Run the following python code:

```python
from can.interfaces import vector
bus = vector.VectorBus(0)
```

### Expected behavior
The bus object shall be initialized correctly.

### Additional context

OS and version: Windows 10
Python version: 3.8.10
python-can version: 4.1.0
python-can interface/s (if applicable): Vector CANcaseXL

The setup works with BusMaster v3.2.2.

The failing test is in `can/interfaces/vector/canlib.py` line 278:

```python
if not all(
    [
        bus_params.bus_type is xldefine.XL_BusTypes.XL_BUS_TYPE_CAN,
        _can.can_op_mode
        & xldefine.XL_CANFD_BusParams_CanOpMode.XL_BUS_PARAMS_CANOPMODE_CAN20,
        _can.bitrate == bitrate if bitrate else True,
    ]
):
```

This test has been added in commit cd51ec4c07041c7b3a15b999aa4be5ec3829b05e.

Maybe it should be extended to account for the case when `_can.can_op_mode` = 0?

<details><summary>Traceback and logs</summary>

```
Traceback (most recent call last):
  File "C:/Program Files/JetBrains/PyCharm Community Edition 2022.2.2/plugins/python-ce/helpers/pydev/pydevd.py", line 1496, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2022.2.2\plugins\python-ce\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:\git\xxx\essai.py", line 6, in <module>
    bus = vector.VectorBus(0)
  File "C:\git\xxx\venv\lib\site-packages\can\util.py", line 337, in wrapper
    return f(*args, **kwargs)
  File "C:\git\xxx\venv\lib\site-packages\can\interfaces\vector\canlib.py", line 286, in __init__
    raise CanInitializationError(
can.exceptions.CanInitializationError: The requested CAN settings could not be set for channel 0. Another application might have set incompatible settings. These are the currently active settings: {'bitrate': 250000, 'sjw': 1, 'tseg1': 10, 'tseg2': 5, 'sam': 1, 'output_mode': <XL_OutputMode.XL_OUTPUT_MODE_NORMAL: 1>, 'can_op_mode': <XL_CANFD_BusParams_CanOpMode.0: 0>}
python-BaseException
```

</details>

zariiii9003, 2022-11-29:
That one is on me. To be fair, that behavior is not documented:
[image]

So there are no other applications active on that bus? Do you have init access? Does it work if you comment out just that one line with the canOpMode check?

Btw, thank you for the well written bug report and analysis.

zariiii9003, 2022-12-16:
@ntstimio Can you check what the value of can_op_mode is, when you are connected via BusMaster?

You could check it like this:
```python3
from can.interfaces.vector import get_channel_configs

for vcc in get_channel_configs():
    print(vcc.name, vcc.bus_params)
```

MGoesswein, 2023-01-12:
I also have a Vector CanCaseXl equipped with two CAN Piggies and have the same issue.

As already written by ntstimio: 
can_op_mode is zero and checked against 1 (xldefine.XL_CANFD_BusParams_CanOpMode.XL_BUS_PARAMS_CANOPMODE_CAN20) in canlib.py, which raises an exception.

[image]

There is no value defined for 0, but it seems that 0 shall also be considered as CAN2.0 compatible. 
Maybe it is only related to older Vector interfaces, because it works if I use a VN7610 interface instead.

**Some Versions as reference:**

**Python:**
Python 3.8.5
python_can-4.1.0

**CanCaseXL:**
Hardware Revision 3.0
Vector Driver/Firmware: 8.7.16 (newest for CanCaseXl)
Internally equipped with 2x CANpiggy 1041Amag (Highspeed), Hardware revision 2

**VN7610:** 
Hardware Revision 2.0
Vector Driver/Firmware: 20.30.14

zariiii9003, 2023-01-12:
Thanks for the confirmation. You say 

> There is no value defined for 0, but it seems that 0 shall also be considered as CAN2.0 compatible.

Does it work with CAN FD? Or is CanCaseXl incompatible to FD?

MGoesswein, 2023-01-12:
> Does it work with CAN FD? Or is CanCaseXl incompatible to FD?

CanCaseXL only supports CAN2.0, no CAN FD. 
This interface is quite old and not sold anymore by vector, but works great for CAN2.0 messages.  Maybe CanOpMode is not implemented (or not correctly implemented) in the driver of CanCaseXL. There was no update of the CanCaseXL driver/firmware in the last years.

zariiii9003, 2023-01-12:
Try #1486 please. I will merge it if it works okay for you.

MGoesswein, 2023-01-12:
Yes, looks good, the exception is gone.

PR #1486 (2023-01-12): Vector: Check sample point instead of tseg & sjw
Also fixes #1453
closed 2023-01-12 by zariiii9003